roman-corgi/corgidrp #1: Estimate star position underneath coronagraph

semaphoreP, 2023-12-15:
Develop routine to estimate the position of the star behind the coronagraph using satellite spots or other techniques. 

1. Requires simulating coronagraphic images (assume they have already been processed to L3 [detector processing and distortion correction is done])
2. Writing a function that finds the star x/y position in each image. Possibly need to develop ways to automatically find the satellite spots in the image. 
3. Write the star position into the header of each image.

m-samland, 2024-02-03:
I would be happy to implement this, but would require some help from someone familiar with the simulations.

semaphoreP, 2024-02-05:
Let's discuss at the meeting, but I think we can use the existing OS11 simulations, but possibly inject artificial satellite spots at the appropriate locations and flux ratios. This can be the mock data until we do full high-fidelity sims.

nasavbailey, 2024-10-30:
@mygouf : can the already-delivered CGI Calibration software simply be released (into the CGI II&T repo) and then ported to DRP? That software fulfils the requirement:

1115530 - Given 1) a clean occulted focal-plane image with a base DM setting, 2) a clean occulted focal-plane image with a relative satellite-spot DM setting added, and 3) a clean occulted focal-plane image with the same relative DM setting satellite-spot subtracted, taken with a given CFAM filter, the CTC GSW shall compute the 2d location of central star and of the spots, in units of EXCAM pixels

mygouf, 2024-11-04:
Yes, we do have code that fulfill this requirement. This task is a code porting activity. However, the code is not yet in the corgi II&T repo. I'll work on releasing it.

semaphoreP, 2024-11-04:
@mygouf is there also simulated data accompanying the code that can be shared? It seems like @m-samland is quite familiar with satellite spot finding codes, but could use some help preparing the data (so that we can properly adapt the code to handle the right data format).

mygouf, 2024-11-04:
Yes, please have a look at this repo for a start: https://github.com/roman-corgi/cgi_iit_drp/tree/main/Calibration_NTR/cal/occastro, there are tests and associated test data. We also have TDD VAP testing code that I will share after we're done with that round of testing.

m-samland, 2025-01-10:
@mygouf I am having trouble reading the fits files in the occastro/testdata folder. Could you check if the data is corrupted?

m-samland, 2025-02-20:
I have made a branch called "star_center" that includes a direct port of the IIT pipeline. It includes a mock function to create test data and a unit test that implements a simple check on whether the known star center is retrieved within a margin of error. The find_star function has been implemented in l3_to_l4, and most of the heavy lifting is done in the module "star_center.py". Currently only the NFOV is added, but it's easy to add the other modes to the dictionary of expected parameters in star_center.py.

closed 2025-03-19 by maxwellmb