roman-corgi/corgidrp #11: Define science data contents

semaphoreP, 2023-12-15:
Currently, science data consists of a primary header, and an image extension with header and data. Do we want to modify this?

Possibly things to add: uncertainty and DQ flag maps. If so, this will need to be created and saved by this pipeline.

maxwellmb, 2024-01-08:
I'm very much in support of uncertainty and DQ flags maps

semaphoreP, 2024-01-08:
L1 data should be (2200, 1200) in (x,y) size. 
L2 and beyond data should be 1024,1024 in size.

maxwellmb, 2024-01-08:
Following the II&T processing the cropping happens from L1-L2a.

maxwellmb, 2024-01-15:
Noting for future reference: The different levels of data have requirements on what the different data Processing steps need to output (sometimes more than just a frame). See the Data Processing Pipelines tables on the JPL Functional Design Descriptions confluence pages. 

For example, one of the requirements is outputting bad pixel maps, so some form of DQ map might be actually formally required anyway..

m-samland, 2024-01-16:
It would be useful to have this, since some steps might flag pixels differently, i.e. we may flag saturated pixels differently than a "normal" bad pixel, as they may be treated differently by a later step.

semaphoreP, 2024-01-16:
Maybe useful to brainstorm some ideas and discuss at the next data telecon!

JuergenSchreiber, 2024-01-17:
Since I promised to implement it, I have some questions:
My first idea was to optionally extend the image class with error and dq attributes additionally to the data attribute.
But I could also make a new class inheriting the image class to keep a simple image class. 
The dq array could be a boolean one, but if we want to distinguish different kind of bad pixels (saturation, cosmics etc) we should have a bit mask and define the different bits.
What header should we use for the error and dq extension? I guess a simple one. The data extension has a long header having a lot of housekeeping keywords. In my opinion these keywords should go to the primary header, the data header should mainly consist of WCS information.

semaphoreP, 2024-01-17:
I think we should extend the image class. Any L1 image read in without DQ and Error in the FITS file should have one created. 

For flexibility, let's use an integer mask for DQ. It can currently be just [0,1] (0 = good, 1 = bad), but if we want more specificity on the DQ flag later on, we can add higher number integers. 

Given that we already have two headers, my vote would be to keep the DQ and error extension headers very simple. I think the division between the primary header and the image extension header right now is that telescope (which we have no control over) populates the primary header and the CGI instrument (which we have some say in) populates the image extension header. Thus, many CGI specific things appear only there.

maxwellmb, 2024-01-30:
I think let's move the discussion of DQ values to Issue #24 to keep it as an explicit issue.

semaphoreP, 2024-01-31:
I'm going to close this issue given we added DQ and error maps in PR #22. We should open specific issues for other data format issues, otherwise this issue will just linger open.

closed 2024-01-31 by semaphoreP