roman-corgi/corgidrp #2: Process data in photon-counting mode

semaphoreP, 2023-12-15:
Produce 2D images in photo-counting mode (L2b images) that can then be fed into the regular astrometry and PSF subtraction codes

- This requires simulating detector data at sufficiently high frame rates
- Implement a photon counting threshold (do we also need to figure out this threshold eventaully)
- Account for photon-counting photometric corrections
- Dark subtraction in photon-counting mode

maxwellmb, 2024-01-15:
If we create error extensions (which I'm a fan of), we'll need to adjust them to account for the added EM Gain noise.

maxwellmb, 2024-01-15:
Note the separate, but related issue #21, which is related to calibrating the EM Gain

kjl0025, 2024-03-05:
This may have already been discussed, but just wanted to document:  https://github.com/roman-corgi/PhotonCount does these things (selects a threshold, accounts for photometric corrections, etc).

semaphoreP, 2025-02-14:
Completed in #277

closed 2025-02-14 by semaphoreP