roman-corgi/corgidrp #4: Develop astrometric calibration functions

semaphoreP, 2023-12-15:
Develop code to measure the platescale, north angle, and distortion solution from a dataset or multiple datasets. 

Requires
- simulating LMC calibration fields
- find and measure position of sources
- compare positions against reference
- measure average platescale and north angle
- measure distortion solution (ask Jason if you need code)
- determine where platescale/north angle values will be stored. determine where distortion solution will be stored

semaphoreP, 2024-01-08:
boresight calibration is top priority

maxwellmb, 2024-01-08:
A lot of work of this has been done already by @hsergi and others. Don't start without talking to him

maxwellmb, 2024-01-08:
A good place to start is also looking at @rzellem's paper: https://ui.adsabs.harvard.edu/abs/2022SPIE12180E..1ZZ/abstract

hsergi, 2024-01-11:
Thanks @maxwellmb. I will double check which material needs unlimited release and which can just be shared. Targeting next Tue 02/16. Feel free to ping me.

semaphoreP, 2024-02-08:
for the boresight calibration, this will be done before the optical alignment is done. We need to talk with Eric Cady regarding how ratty the PSF will be and develop the code to be robust to this. 

Note that the full astrometric calibration post-boresight calibration will be done one a much better PSF. This is the reason the initial boresight calibration sequence is separate from the full astrometric calibration.

semaphoreP, 2024-08-13:
I'm closing this issue now that PR #113 is done, and will make a new issue for fine-precision astrometric solution + disotriton map.

closed 2024-08-13 by semaphoreP