roman-corgi/corgidrp #6: Create core throughput correction calibration files

semaphoreP, 2023-12-15:
- Simulate data scanning a star across the FOV to map the core throughput
- Process data to create a 2-D map of the off-axis throughput for each pixel (values between 0-1)
- Create a library of off-axis PSFs and corresponding separation/PA of each PSF

maxwellmb, 2024-01-08:
A good place to start is reviewing @rzellem's paper: https://ui.adsabs.harvard.edu/abs/2022SPIE12180E..1ZZ/abstract

hsergi, 2024-12-18:
Marie and I have started to work on this and after considering the different tasks, the amount of work to be done, especially by myself, the deliveries, and due dates have come up with the following action plan:

We will first deal with satisfying the CTC requirements 1090881, 1090882, 1090883 and 1090884,  as per the JPL FDD: [https://collaboration.ipac.caltech.edu/display/romancoronagraph/JPL+Functional+Design+Descriptions#JPLFunctionalDesignDescriptions-CalibrateCoreThroughput](https://collaboration.ipac.caltech.edu/display/romancoronagraph/JPL+Functional+Design+Descriptions#JPLFunctionalDesignDescriptions-CalibrateCoreThroughput)
These must be delivered and are necessary to close this issue.

Thus, we will open new issues on GitHub to deal with each of them: Write the functions within a core throughput calibration module, write and document their UTs, submit their PRs, etc until satisfied. Once they are satisfied, we may find that some of the tasks of this issue have been addressed and we can focus on the remaining ones.

In parallel, we will submit a plan to address the tasks in this issue that are not directly addressed by the CTC requirements. The plan will consist of a minimum set of simulated data and steps to complete this issue. As time permits, we may add more complexity.

If you have any questions please iterate on Slack to avoid cluttering GitHub.

hsergi, 2024-12-20:
These are the inks to each CTC req related to CT:

https://github.com/roman-corgi/corgidrp/issues/273
https://github.com/roman-corgi/corgidrp/issues/274
https://github.com/roman-corgi/corgidrp/issues/275
https://github.com/roman-corgi/corgidrp/issues/276

hsergi, 2025-01-31:
For the record: Definition of Core Throughput calibration file in DRP (caldb) as agreed on 1/30/25:

A CoreThroughput calibration file has two main data arrays:
3-d cube of PSF images, i.e, a N1xN1xN array where N1<=1024 is set by a keyword argument, with default value of 1024. The N PSF images are the ones in the CT dataset (1090881 and 1090884)
Nx3 cube that contains N sets of (x,y, CT measurements). The (x,y) are pixel coordinates of the N1xN1xN cube of PSF images wrt the FPAM's center (1090881 and 1090882)
The CoreThroughput calibration file will also include the FPAM, FSAM position during coronagraphic and core throughput observing sequences in units of EXCAM pixels (1090882)
Note: Reqt. 1090883 provides a core throughput map at the some other locations specified by the user (a rectangular grid by default). It also allows one to retrieve the core throughput value at any single location within the core throughput dataset.

hsergi, 2025-01-31:
As for the two last functions:

- Write a function that allows one to pass in a manual (x,y) and obtain the interpolated core throughput correction at that location: Use the function of 1090883 for a single location.

- Write a function that allows one to pass in a manual (x,y) and obtain the interpolated PSF at that location: First method can be simple: FFT or bilinear, which are enough within the TTR5 region (6-9 lam/D). Write it as part of the CoreThroughput class

hsergi, 2025-01-31:
This is my proposal on how to address this issue once 1090881-2-3-4 are closed.

**Simulate data scanning a star across the FOV to map the core throughput**

Input:
- Use the 21x36 OS11 sims as-is
- TVAC pupil image during core throughput calibration

Method:
- Add some Gaussian noise to each noiseless OS11 off-axis PSF (SNR is a parameter, default~150 as expected from calibration)
- Add some Gaussian noise to the TVAC pupil image to generate 3-5 different images

Output:
- Dataset with noisy OS11 data and noisy pupil images.

**Process data to create a 2-D map of the off-axis throughput for each pixel (values between 0-1)**

Use 1090881

**Create a library of off-axis PSFs and corresponding separation/PA of each PSF**

Use 1090884. Note: Agreed with Jason that we can use (x,y) coordinates for now

**Both of the above should go into a single calibration file**

Satisfied with 1090884

hsergi, 2025-02-22:
PR for 1090881, 2, and 4 submitted: [link](https://github.com/roman-corgi/corgidrp/pull/305)

maxwellmb, 2025-03-08:
@hsergi I wrote some interpolation code and tests. Please see [here](https://collaboration.ipac.caltech.edu/display/romancoronagraph/Max%27s+Interpolation+idea).

hsergi, 2025-03-12:
I'm moving the convo about this to 1090883's Issue: [here](https://github.com/roman-corgi/corgidrp/issues/275). Leaving this issue for updates on e2e tests and the status of the 1090881-2-3-4 issues.

hsergi, 2025-03-15:
PR for CT interpolation: https://github.com/roman-corgi/corgidrp/pull/331

hsergi, 2025-03-27:
PSF interpolation approved (https://github.com/roman-corgi/corgidrp/pull/371). Closing this issue. All requests completed.

closed 2025-03-27 by hsergi