roman-corgi/corgidrp #7: Compute flux calibrations

semaphoreP, 2023-12-15:
For each observing block, perform photometric calibration of a standard star behind the ND filter, measure the photometric calibration factor, and apply it to the science data.

- simulate a photometric calibration sequence assuming detector processing and astrometric calibration is done
- develop function to use knowledge of the star's properties to derive the photometric calibration of the image
- save photometric calibration value
- develop separate function that uses the photometric calibration to calibrate science data

JuergenSchreiber, 2023-12-19:
Would be a nice task for me

semaphoreP, 2023-12-19:
Great! I've assigned the issue to you and invited you to the org (which should give you access to this repo)

nasavbailey, 2024-01-08:
@rzellem & @neilzim have been involved in planning flux calibration so far. Tagging them to collaborate.

semaphoreP, 2024-10-29:
@JuergenSchreiber, still interested in this?

JuergenSchreiber, 2024-10-30:
yes, but might need help

semaphoreP, 2024-10-30:
Ok! Anything in particular you think you'd need help with?

rzellem, 2024-11-04:
> @rzellem & @neilzim have been involved in planning flux calibration so far. Tagging them to collaborate.

also tagging @mygouf as she also helped with flux cal :)

rzellem, 2024-11-04:
> yes, but might need help

hi @JuergenSchreiber - @mygouf and I originally developed the absolute flux calibration plan - advised a JPL intern who did the simulations and helped develop the plan - you can find the details of her analysis in her SPIE paper here:

https://ui.adsabs.harvard.edu/abs/2022SPIE12188E..4GP/abstract

I'd also take a look at our CGI Science Calibration plan paper here:

https://ui.adsabs.harvard.edu/abs/2022SPIE12180E..1ZZ/abstract

Lastly, let us know if you have any questions or would like to schedule a tagup over Teams.

JuergenSchreiber, 2024-11-04:
I'll have a look, thanks a lot

rzellem, 2024-11-04:
> I'll have a look, thanks a lot

no problem! feel free to reach out to me on the Roman Slack or via email: robert.t.zellem@nasa.gov

semaphoreP, 2024-11-13:
Feel free to add photutils as a dependency and use it for aperture photometry. I think background subtraction in an annulus is fine (not sure we're expecting much "background"). I think you can assume we know the encircled energy conversion (e.g., even now, we can use simulated PSFs from https://roman.ipac.caltech.edu/sims/Coronagraph_public_images.html#CGI_OS11_report to get us pretty close).

The conversion is from photoelectrons technically. Regarding what unit to convert to, I'm not sure if there's an obvious one. Perhaps @rzellem has thought about this?

rzellem, 2024-11-13:
hi @JuergenSchreiber:

Ultimately, what we’re trying to do is to calibrate the response of our ND filter. We need the ND filter to avoid saturating on bright targets. We already have a notional measurement by our vendor of the spectral response of the ND filter, but we need to check that in flight. Also the ND filter is in a focal plane, so if we get any particle deposition on it, it could manifest as a change in the throughput of the ND filter, which we’d need to correct for. Take a look at the paper that we published in SPIE (I should be the first author) for more details about what we are proposing to do. I’d also look at Lindsey Paine’s SPIE paper. Ultimately, I don’t think the units really matter- what we want to do is create a “correction function” to correct for any deviations from the ND filter’s spectral response measured by the vendor prior to delivery to what the response is now on-sky.

Re: photutils and background subtraction- I’d urge you to use whatever folks are using for extraction for other parts of the DRP. I’d assume you could also just simply use their code/functions for your own analysis.

Not sure what you’re asking with your final question, sorry, but we will be looking at known cal stars- mostly coming from the Hubble CALSPEC catalog.

Hope that helps- please let me know if you have any other questions- happy to chat via Teams, email, or Slack, if that’s easier as well.

semaphoreP, 2024-11-13:
I think we're talking about the absolute flux calibration right now, not the ND filter calibration.

semaphoreP, 2024-11-13:
Thinking about this a bit more, I think the main goal of CGI to do demonstrate point source contrast, so perhaps we should also offer a flux conversion that's more point source friendly? (The only one I can think of is a zero-point magnitude, but it would be nice to move on from magnitudes)

rzellem, 2024-11-13:
@semaphoreP - yes- the only reason we need to do the absolute flux calibration is to ultimately calibrate the ND filter - otherwise, we don’t care about absolute flux calibrations at all (since all of our measurements are in F_planet/F_star space)

JuergenSchreiber, 2024-11-14:
OK, then I try to understand why a relative flux measurement with and without ND filter in each band is not good enough, I guess you need a reference wavelength for each color filter because you want to have the spectral response? 
By the way, is there a possibility to take a spectrum with and without ND filter?
your answer to  photutils and background subtraction: I don't think we have another one already.
flux units: I think a flux density is easiest considering the units of CALSPEC.

semaphoreP, 2024-11-14:
I'm guessing the thinking is that we probably can't use the same star with and without the ND, due the limited dynamic range of the detector? Thus, we can't just do a relative flux measurement with the same star. 

I don't think we want to rely on the coronagraph's limited (and goal-mode) spectroscopic capabilities for flux calibration. 

Yes, Juergen, you will implement the first flux measurement routine of point sources as far as I'm away. Flux density sounds good to me.

JuergenSchreiber, 2024-11-15:
OK, understood, thanks a lot.
Does somebody know whether we have a header keyword that describes the used color filter, like e.g. '3C'.
I think the filter transmission curve files are small enough to store them in corgi. Maybe planned for a calibration file?
The calspec data, I assume the user has it as input for the time being (might work with a selection routine and wget in future).

maxwellmb, 2025-03-12:
@JuergenSchreiber, just to double check, does PR #287 close this issue?

JuergenSchreiber, 2025-03-13:
> [@JuergenSchreiber](https://github.com/JuergenSchreiber), just to double check, does PR [#287](https://github.com/roman-corgi/corgidrp/pull/287) close this issue?

Yes, thanks
I still have to work on #241

closed 2025-03-16 by maxwellmb