roman-corgi/corgidrp #12: Port L1 -> L2 data from JPL II&T

maxwellmb, 2024-01-08:
Many folks at JPL have already done a lot of work on this. Let's not double efforts. 

We will wait for JPL to approve open-sourcing this before really getting to it. 

Included capabilities functions are: 

**L1_to_L2a:**
This will implement the following steps:
         - Separate image from prescan
         - Compute bias from prescan, including offset subtraction, and
           subtract from image
         - Build per-frame cosmic ray masks and remove cosmic rays
         - Correct for nonlinearity

**L2a to L2b:** 
This will implement the following steps:
         - Convert from DN to electrons
         - Divide by EM gain
         - Subtract bias-subtracted, gain-divided master dark in electrons
         - Divide by corresponding flat field
         - Compute per-frame bad-pixel map from fixed bad pixel map (master bad
           pixel map constructed from calibration data such as dark and flat as
           well as other sources of knowledge where present) and additional
           flagged pixels at the frame level if any.

semaphoreP, 2024-01-08:
Should crop down to 1024/1024 at the end of this process

mygouf, 2024-02-02:
Should I go ahead and get started on this?

semaphoreP, 2024-02-02:
I think @m-samland also had some interest in helping with this. @mygouf perhaps you can divide up some of the work to @m-samland so he can help you with this? You could decide to share a branch where you push code for this.

mygouf, 2024-02-02:
Ok, let me start working on it and think about how to divide and conquer. Once we have a good plan for L1 to L2a we can use a similar template for higher levels port and implementation. We can chat more in person next week.

maxwellmb, 2024-02-03:
a natural way to divide and conquer might be to break up all the functions in the ttr5_analog.py into separate issues. What do you think?

mygouf, 2024-02-03:
Yes this is what I had in mind. But wanted to wait for your review before going ahead and starting creating new issues.

mygouf, 2024-02-03:
> a natural way to divide and conquer might be to break up all the functions in the ttr5_analog.py into separate issues. What do you think?

Yes this is what I had in mind. But wanted to wait for your review before going ahead and starting creating new issues.

m-samland, 2024-02-03:
Yes, I am interested in helping out with porting these to the new pipeline to get used to the code base. It would be good to discuss with @mygouf so that we can coordinate!

maxwellmb, 2024-03-08:
Closing this issue since we have split it up into sub-issues.

closed 2024-03-08 by maxwellmb