roman-corgi/corgidrp #13: Port L2a -> L2b data analysis code from JPL II&T

maxwellmb, 2024-01-08:
Many folks at JPL have already done a lot of work on this. Let's not double efforts.

We will wait for JPL to approve open-sourcing this before really getting to it.

Included capabilities functions are:

**L2a to L2b:** 
This will implement the following steps:
         - Convert from DN to electrons
         - Divide by EM gain
         - Subtract bias-subtracted, gain-divided master dark in electrons
         - Divide by corresponding flat field
         - Compute per-frame bad-pixel map from fixed bad pixel map (master bad
           pixel map constructed from calibration data such as dark and flat as
           well as other sources of knowledge where present) and additional
           flagged pixels at the frame level if any.

semaphoreP, 2024-02-02:
@mygouf, yes please! feel free to also recruit others to help if it is convenient.

maxwellmb, 2024-02-02:
For this one and the other one, it might make sense to split these tasks up
into different issues for the different sub-steps. I anticipate we might
have some new people at the meeting next week who might want to get
involved and having smaller chunks might help get others involved.

maxwellmb, 2024-02-14:
I wanted to break this task up more into functions to split up the work for better parallelization and give some porting tasks over to more Junior folks.  At the moment we have the following functions set up in the L2a_to_L2b.py file that could use some attention: `convert_to_electrons`, `cti_correction`, `flat_division`, `correct_bad_pixels`. @mygouf, have you started on any of these specifically? For the ones that haven't been started, I'll create separate issues. There are some missing steps (e.g. "divide by EM gain"), that will also have to be added.

maxwellmb, 2024-03-08:
This issue is now superseded by Issues #48, #49, #50, #51, #52 and #53.

closed 2024-03-08 by maxwellmb