roman-corgi/corgidrp #25: Add image photon noise to error map

semaphoreP, 2024-01-30:
We need a step that computes the photon noise of the science image itself, and add that to the error map

JuergenSchreiber, 2024-02-01:
That looks like an effort I would be happy to contribute. But that will also raise questions about error propagation and how to organize that. For MIRI they have lots of extensions with different kind of errors like flatfield error, read noise and a combined uncertainty.

maxwellmb, 2024-02-01:
Yeah, I was thinking about this too. I'm wary of the proliferation of number of extensions and would also prefer using the 3-D image approach to this. I can see it being worth it to keep the different terms around sometimes, as it can help evaluate what the limiting noise term is.

JuergenSchreiber, 2024-02-06:
The photon noise seems to be just the sqrt of the signal (e/s) (shot noise), is this correct?

maxwellmb, 2024-02-06:
Yup! We may want to combine adding in the photon noise with the step that
converts from DN (ADU) to electrons. Seems like a logical place to this,
right?

semaphoreP, 2024-02-06:
Sounds reasonable. It should just be done before any dark subtraction.

JuergenSchreiber, 2024-02-29:
Should it be done also before prescan_biassub, so in l1_to_l2a.py? I guess not, so a starter in l2a_to_l2b.py?

maxwellmb, 2024-03-01:
I think there is noise associated with measuring and subtracting the bias, but I don't think that it's the same as photon noise.

closed 2024-04-05 by semaphoreP