roman-corgi/corgidrp #29: Update and expand contributing guide

dsavransky, 2024-02-07:
Contributing guide in README (potentially should move to its own page) could use a few more details on individual steps, and currently does not include information on forking.

maxwellmb, 2024-02-07:
I don't doubt that this is true. @dsavransky would you mind providing some more specific suggestions that motivated this comment?

dsavransky, 2024-02-07:
Primarily based on my earlier question about accepting pulls from people not explicitly added to the repo - this requires that they fork first, so we should describe that in the contributor guide.  I'm happy to provide text.

maxwellmb, 2024-03-08:
@dsavransky pinging you on this. Cleaning up old issues.

maxwellmb, 2024-03-11:
Thanks @dsavransky!

closed 2024-03-11 by maxwellmb