roman-corgi/corgidrp #32: Port L1 -> L2a JPL II&T code Part 1: Make a function to process the prescan areas

maxwellmb, 2024-02-08:
This is an off-shoot of issue #12, L1_to_L2a. 

I'm breaking it up since we want these two functions to be a single function. 

This will implement the following steps:
- Separate image from prescan
- Compute bias from prescan, including offset subtraction, and
subtract from image

taichiuyama, 2024-02-09:
matthias, ell, and I are coding, the functoin of prescan_biassub_v2 in detector.py in the prescan_biassub branch
moving to unit test

ell-bogat, 2024-02-09:
Unit tests to write (non-exhaustive):
* both science & engineering given observation type (make sure that the step fails correctly if obs type is not SCI or ENG for now)
* test full frame switch
* make sure the final data shape is correct 
* make sure there are no nans in the output 
* make sure error and dq frames are updated 
* make sure that the individual Image.data and the Dataset.all_data arrays are still linked

closed 2024-03-05 by ell-bogat