roman-corgi/corgidrp #36: Data-class should reflect changes in single frames in all_data attribute

m-samland, 2024-02-09:
Right now changing a single frame in the data class does not update the whole all_data array, they are two seperate entities. Implementation should be changed such that individual changes are reflected in all arrays.

semaphoreP, 2024-02-09:
I had written the code to reflect changes in the all_data array. How are you changing a single frame?

m-samland, 2024-02-09:
Turns out this is a problem only for the prescan bias subtraction since we are cropping the images, which doesn't work as a view into the data. So for this step only we need to overwrite all the arrays in the object. The current implementation should work for everything else so we will address it for this step.