roman-corgi/corgidrp #41: Track each individual error term

semaphoreP, 2024-02-12:
Turn the Error extension into a 3-D datacube, where each slice is the contribution from one souce of noise (e.g., dark subtraction). The final slice should be all of the terms combined in quadrature. The header of the error extension should record which term is filling which index of the 3-D cube. We may need to make sure the error extension header is appropriately being tracked throughout processing. 

There should also be a function in `Image` called something like `add_error_term()` to add additional error terms.

semaphoreP, 2024-02-12:
Based on discussion in #25

maxwellmb, 2024-02-12:
In the end maybe it should always be the fist slice. Then new error slices
just get slapped onto the end. Seems simpler and may where one would expect
to look?

semaphoreP, 2024-02-12:
Sounds good. First slice is the sum of all errors.

JuergenSchreiber, 2024-02-13:
I take it since I already started on these items

semaphoreP, 2024-03-04:
Implemented in PR #45. Thanks!

closed 2024-03-04 by semaphoreP