roman-corgi/corgidrp #42: Current Dark Subtraction Function does not calculate errors or update error header

maxwellmb, 2024-02-12:
To be completed after Issue #41

JuergenSchreiber, 2024-03-04:
To be sure: in detector.create_dark_calib() I guess I need to determine the error from the dataset frames, I think the stdev is fine for the beginning?
To add the error to the Dark frame I have to extend the input parameters by the error, err_hdr in Dark.

semaphoreP, 2024-03-06:
Yes, I think we can use the standard devation. Note that the error on the mean goes as stddev/sqrt(num_frames) assuming noise is independent between frames. 

Ok! Yes, please modify the dark frame `__init__` call as needed!

JuergenSchreiber, 2024-03-11:
I have uploaded a branch darksub_error now to github. I have done it differently now than I first thought, so I look forward to hear your opinion about it. I have done most of the error calculation now inside the dark and dataset class. 
I think the dark is always initiated with the dataset to calculate the dark as input. So I used this to calculated immediately the error inside the class. And instead of looping through the dataset to call the image.add_error_term() I added the function add_error_term to the Dataset class. But this is of course only valid if the error to propagate is the same for all images in the dataset, which is fine for the dark subtraction but I guess also for e.g. background subtraction, flatfielding. But you have a better overview. If you think this is not a good idea, I can easily move the stuff out of the classes, so please tell me.

JuergenSchreiber, 2024-03-11:
By the way, it seems that we have doubled the dark_subtraction function in main: in detector.py and in l2a_to_l2b.py

maxwellmb, 2024-03-11:
My natural instinct would be to put the calculation of dark error `detector.create_dark_calib` function, the same place where the calculation is done. You could just call add_error_term, right? What made you put in in the Dark class? 

I'm a fan of the add_error_term in the Dataset class. Good thinking! 

Also good catch about the doubling of the dark_subtraction function. I thought I had taken it out, but maybe it snuck back in with someone else's merge.

semaphoreP, 2024-03-11:
I agree with Max that having an add_error_term in the Dataset class as a convenience function is good. '

I also agree with Max that it would be best to separate the Data structing code (i.e., `data.py`) from the computing of terms (e.g., the noise map). Thus, I agree with max we should do any calculations of the error in the `create_dark_calib` and `dark_subtraction` functions but use the Dark/Dataset classes for bookkeeping (adding the error to the error map). I think the only change you need to make is to take the error computation out of the `Dark.__init__()` function and put it into `create_dark_calib()`. It'll just be easier to know where to look for this one line of code in the future (rather than having the sift through many lines of bookkeeping code).

semaphoreP, 2024-03-12:
Completed in PR #63. Thanks @JuergenSchreiber!

closed 2024-03-12 by semaphoreP