roman-corgi/corgidrp #48: Port L2a -> L2b JPL II&T code: Convert from DN to electrons

mygouf, 2024-02-21:


maxwellmb, 2024-03-11:
Requires defining a new calibration file type: `KGain`. 

This can be a fits file with only 1 value. Why use a fits file for just one value, you might ask? Because it will allow us to better keep track of metadata.

kjl0025, 2024-03-13:
Just FYI, the k gain and nonlinearity calibration is being worked on by someone on the TVAC team (Guillermo), so that could be useful when it is finished.

maxwellmb, 2024-03-19:
@LisaAltinier brought up DQ flags and we think that this is a good place to flag pixels as being saturated in the DQ hdu.

maxwellmb, 2024-03-19:
We probably want the full well capacity to be in the `constants.py` file.

mygouf, 2024-03-19:
> Requires defining a new calibration file type: `KGain`.
> 
> This can be a fits file with only 1 value. Why use a fits file for just one value, you might ask? Because it will allow us to better keep track of metadata.

Another possibility would be to create a measured kgain parameter in the calibration file parameter that could then be propagated in the higher level science data headers. May be easier to track single values like this that having to open a fits file to retrieve the data? Or we could also do both. In the iit pipeline this value is manually implemented in a config file FYI but it could also be drafted from the fits file. Anyway, having the value written somewhere other than a fits file (either config file or cal/science headers) would be good for quick checks.

mygouf, 2024-03-19:
> @LisaAltinier brought up DQ flags and we think that this is a good place to flag pixels as being saturated in the DQ hdu.

Do we want this feature even earlier in the code, at the L1 to L2a level?

semaphoreP, 2024-03-19:
> Another possibility would be to create a measured kgain parameter in the calibration file parameter that could then be propagated in the higher level science data headers. May be easier to track [...]

One question is here whether we expect the k-gain to ever gain in time? If it'll always be one value (perhaps a value we update as we get better data), then we can keep it hard coded some where. If we are worried about time dependence, we should track it as a FITS file in the caldb.

maxwellmb, 2024-03-19:
If I recall correctly, the calibration strategy at the moment is assume all calibrations will change until we are sure that they don't. @rzellem, can you confirm?

maxwellmb, 2024-03-19:
> > @LisaAltinier brought up DQ flags and we think that this is a good place to flag pixels as being saturated in the DQ hdu.
> 
> Do we want this feature even earlier in the code, at the L1 to L2a level?

@mygouf So I put it in here, since this is the first time we actually have the data in e- and since the full-well is defined in e- we technically don't know what data hits that threshold before this step. Thoughts?

mygouf, 2024-03-19:
> @mygouf So I put it in here, since this is the first time we actually have the data in e- and since the full-well is defined in e- we technically don't know what data hits that threshold before this step. Thoughts?

I guess we can convert the full-well to dn.

mygouf, 2024-03-19:
> One question is here whether we expect the k-gain to ever gain in time? If it'll always be one value (perhaps a value we update as we get better data), then we can keep it hard coded some where. If we are worried about time dependence, we should track it as a FITS file in the caldb.

I'd be in favor of not hardcoding it indeed as we would first need to make sure it won't move. So if we go the fits file route, I would advocate for porting that info to the header for easier access.

rzellem, 2024-03-19:
@maxwellmb - yes, we assume that all cals are time-variant so we will take all cals every campaign until the data tells us otherwise

JuergenSchreiber, 2024-04-25:
eperdn : float (9.55). Electrons per dn conversion factor (detector k gain).

semaphoreP, 2024-06-10:
Closing as this was addressed in PR #80

closed 2024-06-10 by semaphoreP