roman-corgi/corgidrp #49: Port L2a -> L2b JPL II&T code: Divide by EM gain

mygouf, 2024-02-21:


maxwellmb, 2024-03-08:
There is some useful discussion about this in Issue #46 (which I'm closing, to have it superseded by this one).

semaphoreP, 2024-03-11:
Baseline is to simply used the commanded EM gain. Assume this is in the header. See "Data header wishlist" section in "Data Formats" page.

JuergenSchreiber, 2024-03-18:
It is strange, in the cgi_iit_drp code they look for those keywords: 
        if 'EM_GAIN' in hdul[1].header:
            em_gain = hdul[1].header['EM_GAIN']
        if 'CMDGAIN' in hdul[1].header:  
            em_gain = hdul[1].header['CMDGAIN'] , so CMDGAIN would get the preference over EM_GAIN?

maxwellmb, 2024-03-18:
I believe CMDGAIN is the K-gain, not the commanded EM Gain. I've been.
Pushing to include that in the header as well.

JuergenSchreiber, 2024-03-18:
As far as I understand this: The K-gain is the number of electrons per DN (unit e/DN), and the EM_GAIN is the number of electrons activated by a photon (unit e/photons), is this correct?

maxwellmb, 2024-03-18:
The EM gain is more of a electrons-out per electron-in-from-your-pixel as I
understand it (after the electron multiplying process). Your understanding
of K-gain is the same as mine.

JuergenSchreiber, 2024-03-19:
Thanks, I see, it is the electron multiplication gain. So EM gain is number of electrons per photo-electron. Do we plan to have a unit keyword in the header, I don't find it in the data format document?

mygouf, 2024-03-19:
> It is strange, in the cgi_iit_drp code they look for those keywords: if 'EM_GAIN' in hdul[1].header: em_gain = hdul[1].header['EM_GAIN'] if 'CMDGAIN' in hdul[1].header: em_gain = hdul[1].header['CMDGAIN'] , so CMDGAIN would get the preference over EM_GAIN?

EM_GAIN and CMDGAIN are the same thing, commanded EM_GAIN, it is a residual from different codes and I wanted the IIT wrapper to be robust to those two keywords existing in parallel and representing the same thing. From now on we should use the keywords described in that page: https://collaboration.ipac.caltech.edu/display/romancoronagraph/Data+Formats, so we can forget EM_GAIN. We should define a new keyword for the measured em gain but this keyword should be in the header of calibrated data instead of the L1 science header.

mygouf, 2024-03-19:
> Thanks, I see, it is the electron multiplication gain. So EM gain is number of electrons per photo-electron. Do we plan to have a unit keyword in the header, I don't find it in the data format document?

Sure, you can add it to the data header wish list: https://collaboration.ipac.caltech.edu/pages/viewpage.action?spaceKey=romancoronagraph&title=Data+Formats#DataFormats-Dataheaderwishlist

maxwellmb, 2024-03-22:
When the EM gain is measured, it should set EMGAIN_M and EMGAIN_E header keywords to be the measure value and its error.

JuergenSchreiber, 2024-03-25:
So I guess we need an additional function that calculates the EMGAIN and error? It would be good to raise another issue on this. I am a fan of being as modular as possible, that will help when it comes to debugging. So better more small functions than big functions.

semaphoreP, 2024-06-10:
Closing as this was implemented in PR #73

closed 2024-06-10 by semaphoreP