roman-corgi/corgidrp #50: Port L2a -> L2b JPL II&T code:  Subtract master dark

mygouf, 2024-02-21:


maxwellmb, 2024-03-08:
@mygouf, @kjl0025, do we need something special here, or is what we currently have in the `l2a_to_l2b.dark_subtraction()` good enough (pending the update to propagate errors)?

semaphoreP, 2024-03-12:
Hi @JuergenSchreiber, we should probably look through the II&T pipeline for dark creation to see if there is anything more optimal we should be doing than just a simple mean of the dark frames: https://github.com/roman-corgi/cgi_iit_drp/tree/main/Calibration_NTR/cal/calibrate_darks.  Perhaps you can consult with @kjl0025 on slack, since he wrote those functions for II&T.

kjl0025, 2024-03-13:
Sorry for the delayed response.  The plan for the master dark for Roman is to take many stacks of frames, where each stack is taken at a particular exposure time and EM gain.  Essentially, that module you linked (calibrate_darks) takes all these stacks, subtracts the bias, converts from DN to e-, corrects for cosmic rays, averages each stack, and then does a fit for every pixel over the range of averaged frames for the fitting function s = FPN + gain*time*dark current + gain*CIC.  From this, a synthesized master dark can be created for any given exposure time and gain.  And the plan is to use the synthesized master dark in the frame processing pipeline.  I can work on porting that over if you'd like.

semaphoreP, 2024-03-13:
Thanks for the explanation! Ok, just to clarify the output: the final "master dark" calibration will be the fitted coefficients for each pixel to determine the dark current for any arbitrary exposure time? 

For the purpose of the dark creation function, we can ssume that the bias is subtracted, converted to e-, and corrected for cosmic rays (other functions that other folks will write), so all we need to do is to average each stack and then do the fitting. 

I'll ping both of you on slack to discuss coordination.

JuergenSchreiber, 2024-03-20:
I think, we should call this issue differently, like "generation of  master dark calibration file", which is not a pipeline step. The pipeline step dark subtraction we already have, (well, maybe minor updates needed). As far as I understand, Issue #65 should implement a function that generates the master dark (that should be subtracted) out of the coefficients of the calibration file.

maxwellmb, 2024-06-05:
I'm going to close this issue, because I think the subtraction step is done, it's just the generating the right dark that is not, which is covered by issue #65.

closed 2024-06-05 by maxwellmb