roman-corgi/corgidrp #56: find out what OBSTYPEs to use in create_prescan_files() in corgidrp/mocks.py

kjl0025, 2024-03-01:
Alex Greenbaum should know.

maxwellmb, 2024-03-08:
@agreenbaum, should we expect anything other than SCI and ENG (only those two are currently the only ones indicated in the Data Formats page). What about calibration frames, e.g. Darks?

semaphoreP, 2024-03-08:
@maxwellmb, Alex added some other possibilities (not finalized) on the "Data Formats" page.

maxwellmb, 2024-07-25:
@kjl0025, @agreenbaum, @ell-bogat. Are we satisfied with the current list enough to close this issue? If its not breaking anything then perhaps we can add new things as they come up?

kjl0025, 2024-07-25:
Is this the page @semaphoreP was referring to?   In that case, I think we're fine on this and could add as needed.  On my branch (https://github.com/roman-corgi/corgidrp/blob/0640e436acfed0c6a2c4ad7548ceda67fed7f063/corgidrp/detector.py#L98), I have the geometries for the 4 types (SCI, ENG, ENG_EM, ENG_CONV) that were defined in the CTC's proc_cgi_frame repo.  `prescan_biassub` on my branch can handle any of these, and we can adjust the other step functions to be able to handle them if necessary.

maxwellmb, 2024-07-25:
Yes, I had noticed that, I think that's great. Alright, closing the issue for now.

closed 2024-07-25 by maxwellmb