roman-corgi/corgidrp #68: We need a function that can do frame selection

maxwellmb, 2024-03-22:
We don't know how we'll select frames yet, but we still need this function. For now it can just check whether or not a frame is all NANs and if so get rid of it. That way we can get it set up and still test it.

maxwellmb, 2024-03-22:
If a frame is flagged as bad, it should set an IS_BAD header keyword, with the header keyword comment describing the reason it was rejected.

maxwellmb, 2024-06-05:
@nasavbailey can you confirm that for R3.0.1. this is a no-op code since we don't have any selection criteria yet?

nasavbailey, 2024-06-05:
I'd like to see what we can come up with for a beta version. Even if we aren't sure what the final criteria will be, can we define types of data we'd need to pass to enable frame selection, so we can define the function's interface? Your example of more than X% pix have bad DQ flag, where X is TBD, is a good one.  Another example, reject if RMS tip/tilt exceeds some TBD threshold.  Maybe there are other header keywords we'd also want to compare? 

So we want the ability to pass in the name of a keyword, a value, and a comparison type (=, !=, >, ≥, <, ≤, contains, does not contain). Can we define how we'd pass in those criteria, even if we haven't set them yet?

maxwellmb, 2024-06-06:
Another option would be to just code this into the step function as these things arise. In the case you mention, `bad_pixel_percentage` could be a kwarg to the step function, and we could add other kwargs like that (e.g. `max_rms_tiptilt`). Passing in all those things (keyword, value, operator) seems a bit clunky to me, but I'm sure we could figure out a way to do it that way if we wanted.

semaphoreP, 2024-06-06:
Do we need to define every aspect of the interface now? We can just add arguments in the future as we find useful filters. 

In my experience, I rarely useful keywords or DQ flags to throw away frames. I usually use some sort of image metric (correlation coefficient, least squares distance, flux at some separation, satellite spot properties), but it really depends on the data which one is effective. I find it hard to anticipate something that is very empirical in nature.

maxwellmb, 2024-06-06:
Based on a slack conversation with Jason and Vanessa, we will at least include keyword arguments that define a maximum tip/tilt residuals (from a currently TBD header keyword), and maximum % of bad pixels in the frame. We will also want to see if the OVEREXP keyword is set to True.

mygouf, 2024-06-20:
Hi @maxwellmb, I see you modified the milestone for this one. For R3.0.1, this should **not** be a no-op. It should be a function that will cut on user-specified thresholds for our identified criteria (RMS tip, RMS tilt, NOVEREXP, and OVEREXP). It’s OK if we don’t know what the best thresholds for each of those are now, but we do need a function that will cut on each one based on a threshold. Would you revert back to R3.0.1 for the milestone? This is my understanding that this is the current plan and having this issue a R3.0.2 milestone doesn't seem right.

maxwellmb, 2024-06-20:
hey @mygouf, see the comments from Vanessa and about our slack conversation: we've decided that we need to define the interface with a few example keywords so that the interface is defined but not necessarily fully functional. That's why we changed it from R3.0.1 since that is complete.

semaphoreP, 2024-06-21:
Yes, we addressed the R3.0.1 functionality in PR #104. We can close this issue and make a new one if that is better.

nasavbailey, 2024-06-21:
The way Marie described it was my intent. Sorry for my unclear explanation! We can't check off a "shall" as met if it contains no-ops on required features. So if we don't implement any frame selection capabilities, the L2a->L2b requirement is not met. I think this is not a heavy lift b/c it's only doing a header keyword value comparison to mark as good/bad. (BTW: Marie & I in the processing of doing a formal change request to make CTI _correction_ a "should" so that it can be a no-op without jeopardizing the L2b requirement. We can make CTI correction a "should" b/c we have quantitative analysis showing it's not necessary to meet TTR5. Conversely, TTR5 analysis to date has assumed we'd throw out high jitter frames, so frame selection is a required feature.)

maxwellmb, 2024-06-21:
Ok good to know, the previous language of "can we at least define the interface" led us to believe that was the expected at this point. I agree that it's not a big lift and it shouldn't take too long, but it likely won't be done by the June 30th deadline. It shouldn't be too long after that. 

I'm going to move the CTI discussion to the CTI issue #20.

maxwellmb, 2024-06-21:
(FYI, milestone changed back to R3.0.1)!

semaphoreP, 2024-06-21:
Is OVEREXP still expected to be a header keyword? I see it in the IIT header keyword list, but not the new proposed header keyword list.

nasavbailey, 2024-06-21:
I'd assume yes, since it's in TVAC data headers. I think the wiki header page needs a thorough scrub. Is there someone who could take on the task of transcribing the TVAC II&T headers to the wiki page as a starting point? Example data for L1, L2a, and L2b is at https://collaboration.ipac.caltech.edu/display/romancoronagraph/Example+image+frame

PR #122 (2024-07-05): frame selection tests
## Describe your changes

Adds the functionality to the frame selection step function

## Type of change

- New feature (non-breaking change which adds functionality)

## Reference any relevant issues (don't forget the #)

Closes #68 

## Checklist before requesting a review
- [x] I have linted my code
- [x] I have verified that all unit tests pass in a clean environment and added new unit tests, as needed
- [x] I have verified that all docstrings are properly formatted and added new documentation, as needed
closed 2024-07-05 by maxwellmb