roman-corgi/corgidrp #72: Improve Cosmic Ray modeling and rejection

bnemati, 2024-04-05:
This was an issue under https://jira.jpl.nasa.gov/browse/WFIRST-3062

The goal is to use the TVAC data to create more realistic models of the cosmic ray hits, and also to improve cosmic ray masking and rejection in the ground software.

bnemati, 2024-04-05:
The current CGI frame processing code (L1-L2a) masks out the entire remainder of the row. Also the current simulation in emccd_detect makes tails that are notional but need added fidelity to resemble true cosmic ray tails. 

Better simulation on the one hand and better image processing on the other hand can improve both the accuracy of the simulated fringes and the efficiency of the processing. 

Particularly, the cosmic ray tails, after cleanup, can be significantly shortened and the masks can be shrunk accordingly. 

Here is a basic outline of the steps needed:

characterize the cosmic ray characteristics/parameters as a function of the EM gain 
Parenthetically, there will be a difference probably between L2 protons energy loss in the Si vs. ground level muons: these have different dE/dx
As a result, the best characterization is done on orbit, and can be done even when CGI is not primary
Provide Kevin Ludwick with the parameterization so that EMCCD_detect can be accordingly updated
Verify that the cleanup of the CR tails does not in fact hurt the image quality. Previous study showed that 
CR tails have spurious hits and are hence noisier than the rest of the image (hence, even when cleaned up, these are lower quality pixels in that frame)
BUT: statistically, the noise is low enough that, on the whole, the benefit of the cleanup outweigh the costs
we need to confirm this. 
Implement improved cleanup algorithm in proc_cgi_frame
update the CR mask algorithm (simple) in proc_cgi_frame

maxwellmb, 2024-06-05:
See Kevin's comment in #91 as well. I think these are duplicates: we can do better CR rejection and should implement it sometime.

kjl0025, 2024-06-06:
Bijan's issue is related from to what I said in #91 , but not quite the same thing.  That issue was basically a reminder to myself to update the cosmic ray removal code to the most up-to-date version of GSW, which includes better masking due to additional user inputs which can widen the mask of the cosmic head and can shorten the tail mask, but no modeling of the cosmic ray is actually done to determine those parameters.  The masking of cosmic rays using the GSW tool that II&T open-sourced was not sufficient to create good noise maps (arrays of dark current, CIC, and FPN per pixel in the image area) when applied to TVAC data.  It has since been updated and delivered and does lead to good-quality noise maps.

maxwellmb, 2024-06-06:
Sorry @kjl0025! I've re-opened the issue.

kjl0025, 2026-10-01:
Documenting other ideas here.  The median filter method available in corgidrp >= v5.1:  
--Median filter element shape and size could be tweaked to match observed cosmic rays.  
--Also, thresholding could be applied only to the regions on the subtracted frame that are above the first local minimum above the background, whereas currently the threshold is applied to the input frame (and could, for example, mask not only a cosmic above the background but also a PSF above the background).
--Could include an option to cut up the input frame and apply cosmic ray masking separately to each part in case there is a wide range of "background" regions on the frame.