roman-corgi/corgidrp #74: Package architecture modernization

m-samland, 2024-04-10:
The current way the package is structured and installed is not compliant with modern packaging. Here are some general and easy to change points. To make the package future proof we should stick with modern guidelines before the package as grown too large, which should save us trouble later.

1) Package installation should be done with pyproject.toml as setup.py is considered out-dated.
2) The code itself should be in a "src" directory. This is to make sure that when running tests, python always imports the installed package and the local folder by accident. This is to ensure that always the installation is tested. This also make it less annoying to deal with CI and other automation, that otherwise require hacks to account for the folder structure.
3) Adding robust versioning. The version of the package should follow Tags or Releases on github that are matched to specific commits and automatically reflected in the installation.

m-samland, 2024-04-10:
Created feature branch: https://github.com/roman-corgi/corgidrp/tree/packaging_update

I also included an updated CI yaml config, that makes use of the different installation options provided in pyproject.toml (e.g, install -e ".[test]" to install the optional test dependencies, seperate from the actual dependencies).

semaphoreP, 2024-04-10:
It's great to move to .toml setup, but I would like to hold of moving things into a src/ directory. It seems like the flat directory structure is not going away, and the src/ setup doesn't really add anything, except for creating a bunch of merge conflicts in the near future.

m-samland, 2024-04-10:
Oh... I thought the "git mv" command shouldn't cause any catastrophic merge conflicts, as only the pointer to the file location shifts and not the content of the file, which gets diffed, but if you think it will cause a lot of problems we can stash this suggestion for now.

The src layout is my personal preference as it's cleaner and generally more robust (see second and third point in https://packaging.python.org/en/latest/discussions/src-layout-vs-flat-layout/), but I'm not dogmatic about it if it causes issues. In terms of general work-flow it doesn't really make a difference. :)

Let me know if you want me to remake the branch with a subset of the changes. Feedback on the other ideas I tried to implement are welcome too, including CI which I don't know how to test without actually implementing it. The workflow there gets simplified by the pyproject.toml handling the test dependencies.

Ideally, I wanted to get rid of the setup.py entirely, but right now the setup.py makes the database files. I think the general recommendation for this kind of thing would be to not have the install create these files. Instead there could be an init function that the user calls the first time you run the pipeline. The potential problem I see with the current implementation is that depending on the build environment, there may not actually be a home-folder to put these files in, in which case the installation will fail.

semaphoreP, 2024-04-10:
I looked at that link and I don't think the source layout does much practically other than making you click to go one level deeper, and `git mv` isn't as magical as it could be of a command. I think it'll be a headache right now as people are in the middle of development. 

Regarding CI, we will still need to do a fair bit of setup in the Github Actions configuration. There are always going to be build commands specific to a build system that pip can't generalize out. So, I don't think we should be doing too much in the pyproject.toml for testing.

Make a PR to test the CI changes. The tests will automatically run then.

I would say just undo the moving of everything into the source folder.

maxwellmb, 2024-09-13:
Pushing this to R3.0.1., since its a nice-to-have, not a requirement.