roman-corgi/corgidrp #81: when using the Image.copy() function for sub-classes of Image (e.g. Darks) the sub-class type is not copied

maxwellmb, 2024-05-09:
This was brought to my attention by the fact that @JuergenSchreiber created a new copy function for the KGain class in PR #80. 
You can see this like this: 
`import corgidrp.mocks as mocks
import corgidrp.detector as detector

dark_dataset = mocks.create_dark_calib_files()
dark_frame = detector.create_dark_calib(dark_dataset)
dark_frame_copy = dark_frame.copy()
print(type(dark_frame))
print(type(dark_frame_copy))`

Gives the output: 
[image]

Hopefully there's a way we can generalize this without the need for new copy methods for each new calibration datatype.

semaphoreP, 2024-05-09:
unfortunately the best solution is to write a new version of the copy() class, unless we can guarantee that all constructors have the same signature (all `__init__()` functions take the same constructor). There is a hack we can do, but I can't guarantee it'll always work

semaphoreP, 2024-05-14:
Only if that function actually gets used.

maxwellmb, 2024-05-14:
I suppose. It's low priority, but I still think something we want eventually.

maxwellmb, 2024-05-14:
I made a new low priority tag

JuergenSchreiber, 2024-08-01:
If we want to use the copy module, we could save writing copy functions.
I tested it on a kgain instance and it seems to work:
import copy
#reference copy
kgain_copy = copy.copy(kgain)
#deep data copy
kgain_copy = copy.deepcopy(kgain)

What do you think?

semaphoreP, 2024-08-01:
Good idea!

semaphoreP, 2024-10-11:
Sounds good!

maxwellmb, 2024-10-22:
Closed with Merge PR #217

maxwellmb, 2024-10-22:
So that I understand, @JuergenSchreiber, your PR #217, updated the readme, but it doesn't affect existing implementations, right?

JuergenSchreiber, 2024-10-23:
@maxwellmb yes, correct

maxwellmb, 2024-10-25:
Should we just replace the innards of the existing `copy` functions to use `copy.deepcopy()` instead?

semaphoreP, 2024-10-25:
sounds like a good idea to me

JuergenSchreiber, 2024-10-30:
I am not sure whether I understand what you mean. Deleting the existing copy functions? The copy module also works with the existing copy functions. I think the drawback is that it does not update the drp version and time header keywords 
'DRPVERSN" and "DRPCTIME", which the existing copy functions do.

semaphoreP, 2024-10-30:
I think Max might mean take the ~10 other lines of code that involve new object creation and use deepcopy(), but still keep the few extra lines of code that moidfy DRPVERSN and DRPCTIME

JuergenSchreiber, 2024-11-05:
I have created a new branch  copy_replace where I removed the copy functions from the subclasses and used the Image.copy function using the copy module. It seems to work. Could you please check whether we should make a PR out of it?

semaphoreP, 2024-11-05:
Took a glance at it, and it looks good to me. we can iterate details in PR if needed

semaphoreP, 2024-11-07:
Addressed in #248

closed 2024-11-07 by semaphoreP