roman-corgi/corgidrp #84: The add_photon_noise function should check the EM Gain to see if an extra noise term needs to be applied

maxwellmb, 2024-05-14:
When the EM Gain is active (and we're not photon-counting) the shot noise picks up some extra variance, we should account for this in the `add_photon_noise` function.

JuergenSchreiber, 2024-06-13:
@maxwellmb: Do you mean the sqrt(2) factor of excess noise? So, we have to multiplicate the photon noise by sqrt(2) if CMDGAIN > 1? How do we recognize whether we are photon-counting or not? Header keyword?

maxwellmb, 2024-06-14:
Yes, exactly the sqrt(2) factor of excess noise.

> So, we have to multiplicate the photon noise by sqrt(2) if CMDGAIN > 1?
That is my understanding.

> How do we recognize whether we are photon-counting or not? Header keyword?
I think this doesn't depend on us photon-counting, just on us having
CMDGAIN > 1.

closed 2024-07-31 by semaphoreP