roman-corgi/corgidrp #91: cosmic ray masking update

kjl0025, 2024-05-28:
The masking of cosmic rays using the GSW tool for this was not sufficient to create good noise maps (arrays of dark current, CIC, and FPN per pixel in the image area).  It has since been updated and delivered and does lead to good-quality noise maps.  Need to update the cosmic-ray masking function accordingly.

maxwellmb, 2024-06-05:
This is related to Issue #72, so I'm going to close this one and leave that one open.

mygouf, 2024-06-20:
@kjl0025 I see you are still assigned to this issue, are you planning to work on this still or would you need help with this?

kjl0025, 2024-06-20:
Ah yes, I did instinctively assign myself. :) However, I was thinking that once you (or someone) pulls in the latest cosmic ray code on corgi_iit_drp, perhaps Ell could implement the changes since she did the initial port of the cosmic code?  I could do it, but I am running low on funding for this work.

closed 2024-06-30 by semaphoreP