roman-corgi/corgidrp #97: Make calibration file for detector params

semaphoreP, 2024-06-07:
Things like rowreadtime and other detector params that are slowly changed over time should be moved to a calibration file rather than a function, to make it easier to change/deploy the code on CTC. Right now, we decided to use functions, so this work would just be:

 * make a calibration file that handles all of these variables
 * make a function that can easily generate a calibration file given input values for these variables
 * update the install script so that we generate a default calibration file (with pre-launch TVAC numbers) upon install that is tracked in the calibration database, so there is always a value to use
 * update the tests accordingly

maxwellmb, 2024-07-01:
@semaphoreP, is this closed now with PR #106?

semaphoreP, 2024-07-01:
Completed!

closed 2024-07-01 by semaphoreP