roman-corgi/corgidrp #108: Use the ‘applied gain’ header keyword instead of the ‘commanded gain’ header keyword wherever EM gain is used.

mygouf, 2024-06-18:
There should be a small % difference between the commanded and applied EM gain. Thus it would be preferable to use the applied EM gain instead of the commanded gain. 
Please note that the applied EM gain header doesn't exist yet so this issue cannot be implemented yet, but I'm created it to make sure it doesn't fall under the cracks. Marie will work with SSC to make that happen when Jim comes back from vacation.

maxwellmb, 2024-07-25:
@mygouf, can you comment on whether or not this is needed for R3.0.1?

mygouf, 2024-07-25:
Yes, let's plan on having this implemented for R3.0.1.

kjl0025, 2024-07-30:
The CMDGAIN is *supposed* to be the actual applied gain, assuming a good fit to the parametrization function.  In the future, we may implement a methodology of reading off the EM gain from a given frame and assign some new header like EMGAIN_M for that read-off value.  For low-gain frames, the methodology is only possible if a set of illuminated flat-field frames at the same commanded gain and a set at unity gain accompany the observation. For high-gain, the methodology requires a set of dark frames at the same commanded gain accompanying the observation.  

Recommended practice:  We could do something like what is currently done in the `corgidrp.darks.calibrate_darks_lsq` function with a `try` and `except` statement so that even if "EMGAIN_M" for measured EM gain is never implemented, the code doesn't fail:
```
try: # if EM gain measured directly from frame TODO change hdr name if necessary
     EMgain_arr = np.append(EMgain_arr, datasets[i].frames[0].ext_hdr['EMGAIN_M'])
except: # use commanded gain otherwise TODO change hdr name if necessary
     EMgain_arr = np.append(EMgain_arr, datasets[i].frames[0].ext_hdr['CMDGAIN'])
```

hsergi, 2024-07-31:
This has been implemented in calibrate_kgain.py, as well. https://github.com/roman-corgi/corgidrp/pull/116. 

See also https://collaboration.ipac.caltech.edu/display/romancoronagraph/EXCAM+headers+wishlist

maxwellmb, 2024-07-31:
Note: The `EMGAIN_M` keyword that you and Kevin have used should be for the _measured_ EM Gain, not the "actual" EMGAIN, so we still have yet to implement that everywhere.

maxwellmb, 2024-08-21:
Per an email converation with @nasavbailey and others: " I think it would be reasonable to ask SSC to calculate the “applied” gain value from those coefficients and the telemetered DAC value and put that in the L1 FITS headers.", so we will assume we have a header keyword. I put a header keyword request in as `EMGAIN_A`. I think we're ok now to adjust the code to do this.

kjl0025, 2024-08-22:
I could do a quick PR for this if you'd like.  Where the best estimate for EM gain is needed, I could do a try-except statement that selects the first available value in this priority order:  EMGAIN_M, EMGAIN_A, CMDGAIN.  Where a sort according to EM gain is needed, I would only choose from EMGAIN_A and CMDGAIN since EMGAIN_M would not be consistent from frame to frame.

maxwellmb, 2024-08-22:
This would be amazing, thanks for the offer @kjl0025. Please go ahead and do this. I know we'll need this in the primary step function and in a few of the calibration functions as well.

maxwellmb, 2024-08-30:
This should be closed with PR #174

closed 2024-08-30 by maxwellmb