roman-corgi/corgidrp #110: copy method of corgidrp.data.BadPixelMap fails because no `input_dataset`

kjl0025, 2024-06-19:
I think this could be fixed by doing something like:
```
if ext_hdr is not None:
            if input_dataset is None and 'DRPNFILE' not in ext_hdr.keys():
```

Then either `input_dataset` must be present OR the record of the input dataset's filenames must be accounted for in the `ext_hdr`.  

@hsergi , just wanted to document this here.  Please correct me if I am misunderstanding the implementation.

hsergi, 2024-06-20:
@kjl0025: Sure. I _think_ I copied the implementation from KGain when it was still unmerged. I'll do that asap, though it may be later this week, unless it is urgent. I have to work on something else. Let me know!

kjl0025, 2024-06-20:
I think that should be fine.  I think the main goal is to have the main function done by the end of June even if all aspects of the corresponding calibration data class aren't ironed out, but @mygouf can correct me if I'm wrong.  The KGain class in the master branch currently has an issue regarding input_dataset as well (and I made an issue about it and told Jeurgen).  I have an updated implementation in the calibrate_darks branch in the corgidrp.data.Dark class which is widely applicable b/c it has more possible inputs (like err, dq, etc).

hsergi, 2024-06-20:
@kjl0025 Do you mean that you have a patch ready? If so, please submit the PR. If you want me to submit a PR with the change for BadPixelMap, let me know! Ty!

maxwellmb, 2024-07-25:
@kjl0025, @hsergi this issue can be closed, right?

closed 2024-08-12 by semaphoreP