roman-corgi/corgidrp #112: Calibration Algorithm Generation: CTI Pump Trap Data (II&T Port)

maxwellmb, 2024-06-21:
This is a break-up from issue #20 which currently includes the step function and the calibration generation. This part just covers the generation of the calibrations. 

We need to port over the II&T code. I've started and there's a branch [cti_correction](https://github.com/roman-corgi/corgidrp/tree/cti_calibration). This is a requirement for R3.0.1, even if the step function becomes a "should" (i.e. not a strict requirement).

maxwellmb, 2024-06-25:
@kjl0025 says: 
"There are a few minor changes to the tpumpanalysis code that were made after the port to cgi_iit_drp repo.  Those changes are on the cosmic_update_code branch [here](https://github.com/roman-corgi/cgi_iit_drp/tree/cosmic_update_code/Calibration_NTR/cal/tpumpanalysis). The changes are found in example_script.py, tpump_final.py, and ut_tpump_final.py."

maxwellmb, 2024-07-16:
I'm dropping my current working to-do list here mostly to keep track of myself, but also in case other people get involved: 
- [ ] make the input a dataset object. We need to split the data by temperature (header keyword 'EXCAMT' <span style="color:blue">TBC this is the correct keyword</span>.
) and scheme (header keyword 'TPSCHEME')
- [ ] figure out where we get the tpump phase time from. Should it be a L1 header requirement? 
- [ ] Identify what other information that is output by `tpump_analysis` should be saved in headers/extensions
- [ ] Write tests
- [ ] Update slice_section functions based on what is done in the NoiseMaps and Kgain/NL branches

maxwellmb, 2024-07-19:
Tagging the following requirement numbers from the FDD for my reference: 1090879 and 1090880. 

The input data needs to be "clean focal plane images". From what I can tell, "Clean" means L2 data. @mygouf (and maybe @kjl0025), do either you know exactly what steps should/shouldn't be run for the trap pumping data. I'd like to make it clear  in the headers what the input expectation is.

maxwellmb, 2024-07-19:
Some other general questions/thoughts: 
- is illumination_correction needed if we do a flatfield correction?  
- where will the num_pumps input come from? It is the "Number of cycles of pumping performed for each trap-pumped frame."). Will it be in the CPGS file or the header? 
- It looks like the phase time was previously assumed to be in the 'PHASE_T' header keyword. Are we happy with this keyword. It's fine for me. We need it added to the header list.

kjl0025, 2024-07-20:
> Some other general questions/thoughts:
> 
> * is illumination_correction needed if we do a flatfield correction?  **Yes.  For these trap-pumped frames, the flat field is not really applicable.  The trap-pumped frames are taken in TPUMP mode.  Charge is typically injected into a row or rows, and that charge is clocked up and down in rows a bunch of times while the detector is exposed to no light.  There are 4 pumping schemes.**
> * where will the num_pumps input come from? It is the "Number of cycles of pumping performed for each trap-pumped frame."). Will it be in the CPGS file or the header?  **Should be a header.  It was for the actual data from TVAC.**
> * It looks like the phase time was previously assumed to be in the 'PHASE_T' header keyword. Are we happy with this keyword. It's fine for me. We need it added to the header list.  **Sounds good to me.**

As for what should be saved in the output, you may remember the discussion thread with Vanessa and you and me where I listed the outputs that are needed (although I can't seem to find that thread right now.  I thought it was an issue, but I can't find it right now).

maxwellmb, 2024-09-13:
Closing with PR #180

closed 2024-09-13 by maxwellmb