roman-corgi/corgidrp #125: no calibration numbers in DetectorParams

kjl0025, 2024-07-04:
All calibration numbers should come from calibration class instance input to a function rather than from detectorParams (e.g., k gain), and there should be history records for which calibration files were used for a given step function that directly uses the calibration file.

maxwellmb, 2024-07-04:
I think this should be written out more in the README than in an issue. It seems like a more general philosophy than an issue, so its unclear how we'd ever close the issue. I think adding instructions for developers along these lines would be a good idea, perhaps with an example for the DetectorParams file and one counter example (K-gain as you suggested is fine).

semaphoreP, 2024-07-08:
Yeah, this isn't something that is actionable right now, other than removing K-gain from DetectorParams. If you agree, let's close them.

kjl0025, 2024-07-25:
I agree, we can close this.

closed 2024-07-25 by maxwellmb