roman-corgi/corgidrp #128: .corgidrp directory for config made in __init__.py must be deleted in order for an updated directory to be made

kjl0025, 2024-07-04:
Perhaps we should add to the README that a user should delete .corgidrp if a new config is desired upon installation.  Or we could have something in __init__.py to see if necessary keys are present and if not to remake the config?

kjl0025, 2024-07-20:
Sorry I'm just getting to this.  I deleted my .corgidrp to get __init__.py to install a fresh config, so I don't recall the specific error.  But the error was caused by the addition of a key to the config,  and __init__.py didn't make a new config b/c it currently only does that if one doesn't already exist.   I could fix this in another PR soon if you'd like.

semaphoreP, 2024-07-23:
Sure, although I'd put this at low priority until we start releasing the code for public consumption.

semaphoreP, 2024-08-30:
Remaining task: add to README that deleting the .corgidrp directory is a possible troubleshooting solution

PR #177 (2024-08-31): add .corgidrp folder troubleshooting instructions
## Describe your changes

Add troubleshooting for .corgidrp folder

## Type of change

- New feature (non-breaking change which adds functionality)

## Reference any relevant issues (don't forget the #)

Closes #128 

## Checklist before requesting a review
- [x] I have linted my code
- [x] I have verified that all unit tests pass in a clean environment and added new unit tests, as needed
- [x] I have verified that all docstrings are properly formatted and added new documentation, as needed
closed 2024-08-31 by maxwellmb