roman-corgi/corgidrp #135: Do we want to save read noise?

semaphoreP, 2024-07-19:
**Is your feature request related to a problem? Please describe.**
We calculate the read noise during the Kgain calculation. Do we need it anywhere? 

**Describe the solution you'd like**
If we do, we can save it to the Kgain calibration file.

maxwellmb, 2024-07-19:
Yes, I don't see why we wouldn't stick it in the header. 
Technically it should also be the first entry in the PTC, I think, but I do like the idea of having it in the header.

hsergi, 2024-07-31:
calibrate_kgain stores read nosie and its err in the keywords RN and RN_ERR, respectively. See https://github.com/roman-corgi/corgidrp/pull/116

Also, a request for these two new keywords has been added: https://collaboration.ipac.caltech.edu/display/romancoronagraph/EXCAM+headers+wishlist

closed 2024-07-31 by hsergi