roman-corgi/corgidrp #136: Do we want to include actual_gain_arr in some data objects?

hsergi, 2024-07-23:
In both K-Gain and Non-linearity calibration functions, there's an input array with the actual EM gain values (measured vs. commanded) that is used in the functions. Do we want to include this information in the inout Dataset? Currently, the only keyword related to EM gain is `CMDGAIN`, which is associated with the commanded gain.

See also the comment on PR#116: https://github.com/roman-corgi/corgidrp/pull/116#discussion_r1669132879

JuergenSchreiber, 2024-07-24:
Just to understand that right: I see that this is an input array, which is stored in a txt file? Where and how was this generated? I think this should be a calibration file, so that we can convert the commanded EM gain to the actual one, store it in the header and use it?

maxwellmb, 2024-07-25:
@hsergi I think this is a duplicate issue. Please see #108. If you agree I'd like to close this issue.

closed 2024-07-31 by hsergi