roman-corgi/corgidrp #138: Can nonlin_tableTVAC.txt in test_data be replaced by nonlin_sample.csv?

hsergi, 2024-07-23:
nonlin_tableTVAC.txt might be able to replace the existing [nonlin_sample.csv](https://github.com/roman-corgi/corgidrp/blob/main/tests/test_data/nonlin_sample.csv) and associated fits file. This is a nice-to-have not a full requirement in order to keep the number of test_data files as small as possible: https://github.com/roman-corgi/corgidrp/pull/116#pullrequestreview-2163979641:

kjl0025, 2024-07-30:
I just wanted to note that this .txt file is not the latest and greatest nonlin calibration file from TVAC.  Nonetheless, I am fine with what this issue requests.  The nonlin_sample.csv was a vestige for tests from other ported code that used it, but nothing in the DRP currently uses it.

semaphoreP, 2024-09-09:
Fixed in PR #179

closed 2024-09-09 by semaphoreP