roman-corgi/corgidrp #140: L1 header cross-checks

mygouf, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**

Task 1: Check consistency between [provided L1 frames from II&T data](https://collaboration.ipac.caltech.edu/display/romancoronagraph/Example+image+frame) and the [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368) page.

Task 2: Check consistency between [EXCAM image headers (working page)](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803375) on the wiki, and [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368). 

The page [EXCAM image headers (working page)](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803375) should be edited to add any keywords currently missing. The [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368) page is the reference and what is currently implemented for the L1 data and what was implemented for the II&T L2a and L2b data.
 
Task 3: Check that the DRP implementation is consistent with the [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368) page

**Additional context**
Add any other context or screenshots about the feature request here.

maxwellmb, 2024-09-16:
@juliamilton, my impression was that you're working on this and the other header cross-check issues #141 and #142. Is this true? (also tagging @mygouf)

mygouf, 2024-09-17:
Yes Max, you're correct.

maxwellmb, 2024-09-17:
Perfect, thanks for confirming and adding Julia as the assignee!

maxwellmb, 2024-09-20:
@juliamilton, do you mind giving an update on the status of this issue, and issues #141 and #142

maxwellmb, 2024-09-20:
(closed by accident)

juliamilton, 2024-09-23:
The historical list of L1 headers has been checked and updated against the actual headers used in II&T and is captured in [EXCAM image headers (working page)](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803375).

I cannot see the page referenced as [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368), but I was planning to make a new page today under "Data format and volume" that shows the keywords that are included in the upcoming release for each header in L1 DRP products.

maxwellmb, 2024-10-25:
@juliamilton my impression is that this is mostly complete, with the exception of a few one-off issues (e.g. OBSTYPE being replaced by VISTYPE, and some discussion on ASTROM vs. BORESITE). Do you think it would be appropriate to close these issues and then make specific issues for the remaining problems?

closed 2025-03-16 by maxwellmb