roman-corgi/corgidrp #142: L2b header cross-checks

mygouf, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**

Task 1: Check consistency between [provided L2b frames from II&T data](https://collaboration.ipac.caltech.edu/display/romancoronagraph/Example+image+frame) and the [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368) page.

Task 2: Check consistency between [EXCAM image headers (working page)](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803375) on the wiki, and [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368). 

The page [EXCAM image headers (working page)](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803375) should be edited to add any keywords currently missing. The [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368) page is the reference and what is currently implemented for the L1 data and what was implemented for the II&T L2a and L2b data.
 
Task 3: Check that the DRP implementation is consistent with the [II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368) page

**Additional context**
Add any other context or screenshots about the feature request here.

semaphoreP, 2024-08-30:
I'd add that check that the files output by the l1_to_l2b e2e test does have all the necessary headers we expect, and add this to the existing test.

juliamilton, 2024-09-23:
Tasks 1 and 2 are complete. The historical list of L2a headers has been reviewed and updated to match the actual headers used in II&T. This information is now available in the [EXCAM image headers (working page)](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803375).

It seems that the page referenced for Task 3 ([II&T headers keywords](https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=123803368)) no longer exists. I plan to create a new page under "Data format and volume" today, which will show the keywords included for each header in the upcoming release of L2b DRP products.

@maxwellmb @semaphoreP, could you clarify what is meant by the "list of all necessary headers" expected for inclusion in this release? Additionally, will the constants.py file mentioned in the linked issue be used? Otherwise, I’ll verify the outputs from the L2b end-to-end test to produde the list of headers currently included and flag any missing II&T keywords.

maxwellmb, 2024-09-23:
I don't think we've yet implemented a constants.py. We were hoping to have a list of the required header keywords for each level saved somewhere in the pipeline, such that it could be checked when updating the data level and saving. If not present I think we'd want to create them and populate them with dummy values for archiving purposes. 

Given the current time-frame I think this will have to be bumped to the next release.

closed 2025-03-16 by maxwellmb