roman-corgi/corgidrp #145: K-Gain and Non-linearity calibration need a new value of OBSTYPE

hsergi, 2024-07-25:
As a reminder, these calibration fucntions need a set of independent frames that are used to generate a mean frame. These frames are not a subset of the frames used to calibrate the data. OBSTYPE has NONLIN as one of its values. I added a new value MNFRAME to be able to distinguish both sets in the single dataset that is used as the inout for these calibration functions.

This value, or a different Keyord/Value has to be created in the pipeline.

hsergi, 2024-07-31:
I have added the request in https://collaboration.ipac.caltech.edu/display/romancoronagraph/EXCAM+headers+wishlist

JuergenSchreiber, 2024-08-14:
@hsergi, I guess the MNFRAME keyword is for the files with the same exp time to determine the noise?

maxwellmb, 2024-08-30:
With these header keywords requested on confluence, there's nothing to be done on the DRP side.

closed 2024-08-30 by maxwellmb