roman-corgi/corgidrp #147: Create recipe template for generating a NoiseMaps Calibration

maxwellmb, 2024-07-26:
**Is your feature request related to a problem? Please describe.**

No recipe template exists for creating a NoiseMaps calibration from L1 data. 

**Describe the solution you'd like**

A recipe template needs to be written to generate a NoiseMaps Calibration file from L1 data. 

**Additional context**

In order to test this you will need to get your hands on the appropriate input data from II&T or TVAC and you will quantitatively compare the calibration file to the outputs of the II&T data pipeline. In addition please describe in your own words (~several sentences to a paragraph) what the recipe does and what the inputs and outputs are. Finally, please make sure the appropriate table entry is complete within the Confluence DRP Implementation Document table in Section 2. 

Here's a checklist: 
- [] Recipe Written
- [] Tests Written and passing
- [] Descriptive Paragraph Complete
- [] Table Entry Complete

kjl0025, 2024-08-05:
Some more info on the TVAC testing:  The II&T function for the noise maps at the time of delivery did not handle cosmic rays in a way that allowed for noise maps without pock marks left over from cosmic removal.  That code was updated, but the TVAC analysis person (Guillermo in this case) delivered the maps before this update was done to the II&T code, and he got good results using something other than the official II&T code due to delivery date constraints, and that code is documented.  So the results in the Box folder are good, and running data through the DRP *should* give similar results.  I have confirmed that the results are at least visually similar using the corrected II&T code, which is what was ported into the DRP.

The TVAC results for the 3 noise maps in the Box folder are the *_map_*.fits files in the "results" folder.

maxwellmb, 2024-09-20:
closed with PR #182

closed 2024-09-20 by maxwellmb