roman-corgi/corgidrp #149: Create recipe template for generating a BadPixelMap Calibration

maxwellmb, 2024-07-26:
**Is your feature request related to a problem? Please describe.**

No recipe template exists for creating BadPixelMap from L1 data. 

**Describe the solution you'd like**

A recipe template needs to be written to generate a BadPixelMap Calibration file from L1 data. 

**Additional context**

In order to test this you will need to get your hands on the appropriate input data from II&T or TVAC and you will quantitatively compare the calibration file to the outputs of the II&T data pipeline. In addition please describe in your own words (~several sentences to a paragraph) what the recipe does and what the inputs and outputs are. Finally, please make sure the appropriate table entry is complete within the Confluence DRP Implementation Document table in Section 2. 

Here's a checklist: 
- [] Recipe Written
- [] Tests Written and passing
- [] Descriptive Paragraph Complete
- [] Table Entry Complete

juliamilton, 2024-08-19:
I've written the recipe and test file for this, but have a few outstanding questions:
- How should walker.py handle the master dark and master flat? I've added additional arguments to the walk_corgidrp function to accept these inputs for now, but I'd like to confirm if this approach is what you had in mind.
- The TVAC noise maps I have access to have multiple fields swapped between what the code expects to find in the primary and extension header, which causes several issues. Additionally, KGAIN is not defined in either of the headers for the TVAC images. I've made modifications to darks.py and the Image class to address the swapped header values, but I'm wondering if this is specific to the data files I have or if you’ve encountered similar issues with the datasets you’re working with.

semaphoreP, 2024-08-19:
The master dark and master flat should be calibration files specified as part of the recipe json file. They should be specified as part of the bad pixel map creation step (see the other step functions as examples). You shouldn't be modifying the walk_corgidrp function.

I've not encountered such issues. The code I pointed out in the `l1_to_l2b_e2e.py` should work just fine to handle all of the calibration data you need. Have you gotten access to the box folder yet?

juliamilton, 2024-08-19:
Got it, thanks. I reverted walk_corgidrp to its original arguments. For that, can I make the assumption that only one Dark and one FlatField calibration file will be in the caldb at a time?

Ok, good to know. It may just be an issue with the data I have then, since I did have to modify a lot to handle the noise maps. I don't have access to Box yet - since the invite was sent to my jpl.nasa.gov email address I have to get NASA approval for an account.

semaphoreP, 2024-08-19:
If you set the calibration files to be chosen automatically in the recipe (with the "AUTOMATIC" value), then the caldb will automatically pick the most suitable calibration, even if there are multiple files in the caldb. 

If we can share the Box folder with a different email, just DM me.

closed 2024-09-03 by semaphoreP