roman-corgi/corgidrp #176: ability for desmear step function to determine if desmearing is worthwhile based on inputs

kjl0025, 2024-08-30:
**Is your feature request related to a problem? Please describe.**
For calibrating a synthetic master dark, some frames in the input dataset may need to be desmeared while others may not.  An edit to the structure of the walker could be done to handle this, or we could add optional inputs to the desmear step function to handle this.  This would also be useful for a regular (non-calibration) dataset as well; instead of having the user separately figure out if desmearing is appropriate before specifying a recipe for the dataset, the step function could handle it automatically.

**Describe the solution you'd like**
There could be inputs for expected values, in e- units, for mean FPN (or perhaps max value for FPN since it may have a mean of 0), read noise, mean CIC, and mean dark current.  If these are not specified, they could be taken from a specified synthesized master dark if appropriate.  (In the context of a recipe for making a synthesized master dark, the user would specify the numbers in the aforementioned inputs.)  If none of these are specified, the step function makes no recommendation.  Based on these inputs and the EM gain and exposure time used for a given frame, the step function would determine how far above the values to be subtracted in the desmearing are compared to the non-smeared noise.  If significantly above (determined by an optional input, number of standard deviations which could default to 2 maybe), the desmearing is recommended.  Then there could be an override flag (True or False) to override the step function's recommendation to smear or not.  

**Additional context**
If there are cases where a user wants to override the desmear step function's recommendations for only a subset of frames in a dataset, then I think a change to the walker's structure would be needed.  This would also be true for any other step function for which a user wants different keyword arguments per frame in an input dataset.

semaphoreP, 2024-08-30:
Is this something you want to try to get in this release? Or is this a longer-term project?

kjl0025, 2024-08-30:
It's not necessary for this release.  Just a longer-term project.

kjl0025, 2025-01-23:
Additional notes:
Any component linearly proportional to exposure time will have a smear to it (i.e., photon flux and dark current), while other components will not (CIC, FPN, read noise).

**synthetic master dark:**  
1.  It is made from a mix of frames at various exposure times, so in theory, one should apply desmearing (if appropriate, which is unlikely since the dark current is typically <= 1e-3 e- per pixel, unless the exposure time is large enough) to the stacks before constructing the synthetic master dark.  
2. For a given illuminated set with exposure time t, smear should be APPLIED to the dark current map according to t before synthetic master dark is constructed and subtracted since the illuminated frame also contains this smeared dark current in it.

**traditional master dark (analog and PC):**
1. No desmear should be applied to the dark frames used to make the master dark because the smeared dark current is present in both the illuminated and dark frames for the same exposure time.