roman-corgi/corgidrp #178: allow calibrate_darks_lsq to have sets for unique gain-exposure pairs have different numbers of frames

kjl0025, 2024-08-31:
**Is your feature request related to a problem? Please describe.**
Currently, calibrate_darks_lsq requires an input dataset comprised of smaller sets, each corresponding to a unique combination of EM gain and exposure time and all having the same number of frames.  Running `frame_select` in general will break the latter condition, which will cause calibrate_darks_lsq to crash with an exception.

**Describe the solution you'd like**
Either reduce the number of frames each set by the minimum amount per set so that they all have the same number of frames, or change the fit so that each data point per pixel is weighted by the number of frames in the corresponding set.

**Additional context**
Not needed for the 9/9 release.

kjl0025, 2024-08-31:
Actually, after looking at `frame_select` more, I think it shouldn't be invoked for the recipe to make a synthesized master dark.  The enhancement to `calibrate_darks_lsq` would be nice to have in general in case something goes wrong and CGI ends up taking frame sets with different numbers of frames, but practically, that doesn't seem like that would be an issue.

semaphoreP, 2025-07-18:
Added in #474

closed 2025-07-18 by semaphoreP