roman-corgi/corgidrp #183: Create Recipe Template to make a master dark from Dark Frames

maxwellmb, 2024-09-04:
**Is your feature request related to a problem? Please describe.**

No recipe template exists for creating a Master Dark from a series dark frames. This is different from creating a synthetic dark. 

**Describe the solution you'd like**

A recipe template needs to be written to generate a Dark Calibration file. 

**Additional context**

This is a DRP requirement, but not the expected default behavior.

maxwellmb, 2024-09-16:
@kjl0025 my understanding is that you are to use the data [here](https://ucsb.app.box.com/folder/277970344084) to create the dark_current map [here](https://ucsb.app.box.com/folder/278281799909). @mygouf can you confirm? 

@kjl0025 Do you know where to find this specific II&T/TVAC code in case there are any problems? @mygouf you mentioned that "Dark Calibration" had been updated since TVAC. Is this the code to create synthetic darks, or traditional darks?

kjl0025, 2024-09-18:
Thanks, Max.  And since there is no official II&T for building a traditional dark (e.g., taking mean of a bunch of darks with same EM gain and exposure time), I will simply check that the output of the DRP's `build_trad_dark` is close to the TVAC result for the dark current, which was formed by basically doing this process on the raw frames, which all had the same exposure time and EM gain.  Correct me if I'm wrong on any of this, @mygouf .

maxwellmb, 2024-10-03:
Closed with PR #205

closed 2024-10-03 by maxwellmb