roman-corgi/corgidrp #194: Local file writes will fail when the repo is used in the official GSW docker image

ajeldorado, 2024-09-10:
**Describe the bug**
In various files, there are local file writes for directories such as with `os.mkdir` and for files such as with `plt.savefig`. The repo will eventually be used in a docker image that does not allow writing files within the repo, so the unit test suite will fail in that situation. In the other GSW repos, we have used `tempfile.mkstemp()` to avoid this problem.

**To Reproduce**
Steps to reproduce the behavior:
This behavior cannot be replicated until after the code has been delivered and packaged into a docker image.

**Expected behavior**
The unit tests are all expected to pass.

**corgidrp version:**
 - version 0.2.1

**Additional context**

semaphoreP, 2024-09-10:
When we delivered the interim release of corgidrp to the CTC, the unit tests ran and we didn't have this issue. What would be different in a future release?

ajeldorado, 2024-09-10:
OK, that's good then. Going through the PRs and Issues for the other GSW repos, it looks I had the problem stated wrong. The file writing worked OK, but we had a problem with the files written during the unit test persisting. So the solution there was to make sure to delete the created files once they were done. That was likely a problem with the tests that wrote hundreds or thousands of full sized frames. If the unit tests in corgidrp only write a handful of small files, then it might be OK to leave them as is.

semaphoreP, 2024-09-10:
Ok, closing the issue for now then. Please re-open if needed.