roman-corgi/corgidrp #203: detect_cosmic_rays draws k gain from DetectorParams instead of KGain class

kjl0025, 2024-09-18:
**Is your feature request related to a problem? Please describe.**
Currently, `detect_cosmic_rays` draws the k gain value from `DetectorParams`, but k gain is a calibrated value coming from the corgidrp.data.KGain class.  

**Describe the solution you'd like**
Change the function to draw k gain from corgidrp.data.KGain instead, and adjust tests accordingly.

**Additional context**

JuergenSchreiber, 2024-09-24:
Should we make it optional and leave the default value in DetectorParams, if KGain cal file not available. Or should we delete the kgain value in DetectorParams?

kjl0025, 2024-09-26:
Sorry I didn't get back to you sooner on this!  I shared my thoughts on your PR.

semaphoreP, 2024-10-22:
Added in PR #211

closed 2024-10-22 by semaphoreP