roman-corgi/corgidrp #223: Noise map inputs must be 1024x1024, not 1200x2200. Fix needed immediately

ajeldorado, 2024-10-22:
**Describe the bug**
corgidrp.darks.build_synthesized_dark() errors when given 1024x1024 arrays in the noisemap stack. `TypeError: F must be 1200x2200`

**To Reproduce**
Try running corgidrp.darks.build_synthesized_dark() with 1024x1024 input noise maps.

**Expected behavior**
corgidrp.darks.build_synthesized_dark() needs to have the same basic functionality of the deprecated function it replaces, cal.masterdark.gsw_masterdark.build_dark(). The input maps for build_dark() were and should all be 1024x1024 arrays because that is what all the use cases need in FSW and GSW.

**corgidrp version:**
v1.0

**Additional context**
**This needs to be fixed immediately** and included in a bug fix release (v1.0.1 ?) for GSW R3.0.1 testing.

I recommend correcting the array size checks in corgidrp.darks.build_synthesized_dark(), changing corgidrp.data.DetectorNoiseMaps to use input 1024x1024 maps (and removing the `full_frame` option), and then fixing anywhere else that breaks from that change.

semaphoreP, 2024-10-22:
@ajeldorado Why does the input need to be 1024x1024? Shouldn't we only care that the output is 1024x1024? Is there a requirement that the noise maps are 1024x1024

ajeldorado, 2024-10-22:
- The main reason is that the F, D, and C frames are also meant to be used in FSW, which can accept only 1024x1024 frames for each of them.
  - While it could be different for GSW and FSW, then we have to bookkeep two different versions which is problematic.
- A 1200x2200 master dark isn't usable for anything because L2a images already have to be 1024x1024.
- Dark current noise in the overscan is not sensible.

ajeldorado, 2024-10-22:
I'll counter with asking, why do the inputs need to be the current 1200x2200? They are processed data products anyway and don't need to be the same size as the frames they were created from, the info outside the 1024x1024 image area is never used, and the agreement was to port over the functionality of the existing code that takes 1024x1024 inputs and creates a 1024x1024 output.

kjl0025, 2024-10-22:
The II&T code for making noise maps (calibrate_darks_lsq) made a full-frame master dark, and this was because the EM gain measurement code (EM_gain_fit from cal repo) made use of the full-frame master dark (and the function enforces 0 dark current in non-image areas).  

Could we compromise and save the calibration noise map files with a 1024x1024 HDU and another full-frame HDU?

semaphoreP, 2024-10-22:
I'll note that we can use keywords in `corgidrp.darks.build_synthesized_dark()` to input 1024x1024 noise maps without crashing. 

It sounds like we do need the full frame 1200x2200 noise maps for DRP calibrations. Can you just crop down the 1200x2200 into 1024x1024 as part of reading in the DRP-produced FITS files?

maxwellmb, 2024-10-22:
I'll say that we've gotten no direction or requirements from the data
formats for FSW.
I'd imagine we might want different recipes for FSW products and that might
resolve this too, no? e.g. with different detection_regions.

semaphoreP, 2024-10-22:
Yes, we can certainly produce 1024x1024 noise maps by just changing the input recipe into the DRP

ecady-jpl, 2024-10-22:
Give me a minute and I can talk to FSW needs...

ecady-jpl, 2024-10-22:
OK, FSW will need the following 5 things at 1024x1024 (across everything, not just darks):

- FPN array
- CIC array 
- dark current noise array
- flat field array
- bad pixel array

ecady-jpl, 2024-10-22:
Flat field is new, clearly.  Bad pixels and 3x synthetic dark components were moved over from WFSC.

If there is a path to getting those 5, I think we can make this work.

maxwellmb, 2024-10-22:
Two questions:
- Are there other general FSW requirements that we should know about?
- Are these new requirements necessary for acceptance of R3.0.1?

ecady-jpl, 2024-10-22:
And to address @ajeldorado's original concern: the standing VAP functional tests load in FSW-sized arrays.  Is there a native way in `corgidrp` to embed an image-sized array in a full frame?

We can throw something together with tools from another repo, or rebuild the test data maybe, but it would be preferred to use `corgidrp` tools if available.

ecady-jpl, 2024-10-22:
@maxwellmb 

- This is a requirement in that FSW has allocated a memory region for each of these and the data needs to be the right size for the buffer, and it won't be allowed to be uploaded if it's the wrong size.  There is a very thin L1 -> L2a -> L2b pipeline onboard and it requires a subset of the products used in corgidrp.
- There are also a couple of scalars that will need to be extracted from tools now in corgidrp: K-gain and bias offset
- As long as there is a path to producing these products with your tools, we should not have to hold up R3.0.1.

maxwellmb, 2024-10-22:
Eric, thanks for the reply. I'm also more generally wondering if there are
formal requirements written down anywhere for what the FSW needs from the
DRP just to make sure we don't miss anything else and can incorporate it
into future tests.

There's a chance there will be some debugging, but there's nothing
fundamental about the DRP that wouldn't allow this. In fact using the
detector_regions keyword we can probably resolve this without changing
code, and only changing a recipe.

ajeldorado, 2024-10-22:
That brings up another issue...should I make a formal GitHub Issue for it?
We can't directly access things in the docker image containing corgidrp that are data types, only functions, classes, and methods. So that means we can't just load `corgidrp.detector.detector_areas` because it is a dictionary--it would need a very thin function wrapper that returns the dictionary for us to access it. There may be other cases like this in the other modules too.

semaphoreP, 2024-10-22:
Ok, so it sounds like we should be outputting 1024x1024 noise maps for FSW. From my reading of the discussion above, it's ok if we have to alter the default parameters and use a specialized recipe to produce these 1024x1024 noise maps. Please confirm this is ok, and we will write a test that checks this functionality works. 

I believe flat field and bad pixel maps are already 1024x1024 by default, so this is the only one that needs adjusted. 

@ajeldorado I don't understand why you can't access `corgidrp.detector.detector_areas`. Could you give some more details? In python, everything is an object (including functions/classes/methods), so you can do `import corgidrp.detector; print(corgidrp.detector.detector_areas`. And docker itself is fairly agnostic to exact python behavior. So presumably there is some intermediate interface you are dealing with? If so, we're happy to write a wrapper function if that's helpfu..

maxwellmb, 2024-10-22:
Also, do be clear: you shouldn't have to modify
corgidrp.detector.detector_areas,
most of the places where you might slice up the detector in different ways
can be accessed by passing in a different detector_regions keyword that
supersedes detector_areas. Is that what you want to do?

ajeldorado, 2024-10-22:
@semaphoreP Yeah, there's another, custom interface layer from Al Niessner called DORBA that we use, and that is the part that can't handle anything other than functions, classes, or methods. DORBA is helpful in that it deals with all the XML and pickle file nonsense of the docker image interface to make it look like and feel like you are using the Python code under the hood of the docker image without the docker image. But there are some differences, like that, and also all function or method inputs via DORBA are keyword arguments, which is annoying.

For this instant, I'd want something like `detector.imaging_area_geom('SCI')` that spits out 1024, 1024, [13, 1088] , but it actually gives back the larger area 1037 1056 [0, 1088]. So for lack of more specific functions (unless they are there somewhere), I'd like to be able to retrieve the contents of corgidrp.detector.detector_areas from the docker image rather than having to hard-code values copy-pasted from the GitHub repo. So the point is not to change the values, but to access them at all from the docker image hidden behind DORBA.

ecady-jpl, 2024-10-22:
> Eric, thanks for the reply. I'm also more generally wondering if there are
formal requirements written down anywhere for what the FSW needs from the
DRP just to make sure we don't miss anything else and can incorporate it
into future tests.

@maxwellmb maybe we take this to email?  There's a lot of material in the FDDs which documents the formal design, and quite a bit is assigned to GSW; the decomposition further into DRP vs WFSC is not defined there though.  But we can probably scrub the requirement list and cross-check.

> Ok, so it sounds like we should be outputting 1024x1024 noise maps for FSW. From my reading of the discussion above, it's ok if we have to alter the default parameters and use a specialized recipe [...]

@semaphoreP, that should be fine as long as it's documented and we can access it via DORBA.

fwiw, @kjl0025's code [already makes the 1024x1024 maps](https://github.com/roman-corgi/corgidrp/blob/main/corgidrp/darks.py#L626-L628), but they didn't make it to `noise_maps` output in the port.

semaphoreP, 2024-10-22:
> For this instant, I'd want something like `detector.imaging_area_geom('SCI')` that spits out 1024, 1024, [13, 1088] , but it actually gives back the larger area 1037 1056 [0, 1088]. So for lack of [...]

@maxwellmb @kjl0025 could one of you briefly check that none of the current detector.py helper functions do the thing requested here?

kjl0025, 2024-10-23:
`unpack_geom('SCI', 'image', detector_regions)` gets you 1024, 1024, [13,1088], and `slice_section(frame, 'SCI', 'image', detector_regions)` returns the actual 1024x1024 subframe.  You can do `from corgidrp.detector import unpack_geom, slice_section` to get those.  (The other one, `imaging_area_geom()`, is slightly bigger than 1024x1024 b/c it includes shielded pixels.)

Will that suffice for building the synthesized dark (embed the image area into a full frame)?  Or do I need to make  `corgidrp.darks.build_synthesized_dark()` have the option to allow for 1024x1024 inputs?

semaphoreP, 2024-10-23:
@ajeldorado Are you intending on using the `unpack_geom()` to carve out the necessary data you want, and that solves the array size mismatch issue? Or is that separate, and we should still be producing 1024x1024 noise maps and darks (by modifying the functions in the DRP)? If it's the latter, is it ok we are producing this on demand, and not as the default noise map processing?

ajeldorado, 2024-10-25:
I think Eric and I agree now that we don't need 1024x1024 noise maps produced at the same time, so the full size frames are OK. 

What would be good is to have two new utility functions somewhere to extract the 1024x1024 array out of the 1200x2200 array or do the reverse and place a 1024x1024 array into the correct location in the full 1200x2200 frame. I've noticed that this is done by direct array indexing in the test files, but we'd really prefer to avoid that so we don't get really dumb errors like getting x/y-row/col mixups, 0 vs 1 indexing mixups, or similar, and so that someone doesn't have to keep looking up how to do that (also a potential source of error).

semaphoreP, 2024-10-25:
It seems to me this is what `slice_section()` does that Kevin mentioned above.

ajeldorado, 2024-10-25:
`slice_section()` works for grabbing the subset of a full frame, but it does not appear to put the cleaned frame into a full frame, which is what I need for the reverse operation.

kjl0025, 2024-10-25:
I should be back to work on finishing my PR #215 by Monday, so I can slip in the embedding utility function next to `slice_section()`.  Unless you need it sooner, @ajeldorado .

ajeldorado, 2024-10-25:
If it can go in the same bug fix release as #229 (which is urgent and immediate), then that would be nice to have but not required.

FYI, I'll be on vacation October 28-November 8.

semaphoreP, 2024-10-26:
Happy to wait for both PRs to be done to review. Seems like both are already in good shape, so shouldn't be long.

semaphoreP, 2024-11-04:
The ability to embed has been added in PR #215  (thanks @kjl0025!). See new function `corgidrp.detector.embed()`. @ajeldorado let us know if this satisfies this issue, and if so, we can close this.

ecady-jpl, 2024-11-05:
@ajeldorado is on vacation but he asked me to keep things moving forward while he's out.  It looks like `embed(data, 'SCI', 'image')` should get him what he needs.

ecady-jpl, 2024-11-05:
And as a follow-up: is it possible to get a point release to Cynthia/Al now that #215 and #229 are complete and merged?  Will still need a couple days for them to make the new docker image.

semaphoreP, 2024-11-05:
Goal is a release by the end of the week.

maxwellmb, 2024-12-13:
@ajeldorado I think this issue can be closed, right?

ajeldorado, 2024-12-13:
Yes, OK to close now.

maxwellmb, 2024-12-13:
Thanks @ajeldorado!

closed 2024-12-13 by maxwellmb