roman-corgi/corgidrp #227: No .corgidrp folder

Post169, 2024-10-23:
I installed corgidrp on my Windows 10 computer, tried to run l1_to_l2a_e2e.py, and it crashed because there is no .corgidrp directory, let alone files within it, in my home directory.

I installed most dependencies listed in requirements.txt using Anaconda (e.g., `conda install numpy`), and then executed `pip install -e .` to get the rest. There was no report of a problem on the Anaconda Prompt.

After reading the Readme, I saw that there should be a .corgidrp folder in my home directory, checked, and saw none, so I knew something was wrong.

I downloaded the main branch on Tuesday, October 23. I think Jason Wang's "Merge pull request #211 from roman-corgi/cosmics_kgain" (9489f00fa1ed05b6469506e4446ac8c6e4a7fada) was the last commit when I installed.

semaphoreP, 2024-10-23:
Can you print out what you get for the following? `import corgidrp; print(corgidrp.config_filepath)` and confirm that file does not exist? 

We've never formally tested corgidrp on windows, so you may need to use Windows Subsystem for Linux.

Post169, 2024-10-25:
It looks like the .corgidrp folder was created when I ran the script in Windows, but only with an empty `default_calibs/` folder and a `corgidrp.cfg` file, nothing else.
I made double sure to install git lfs ahead of time

I set up WSL and ran corgidrp from a completely native installation, and it looks like everything worked there.
As I learn the package more, I'll try and find clues as to why it crashes in Windows.

A peripheral detail - the Readme states that the ~/.corgidrp directory is created upon installation of the package, but even on WSL it only appeared when I ran a script for the first time.

semaphoreP, 2024-10-25:
thanks, that used to be the behavior, but we changed it (for pip-related reasons) to create .corgidrp on install.

semaphoreP, 2024-10-25:
regarding your windows install, try deleting the whole .corgidrp folder and try again.

Post169, 2024-10-28:
Deleted every trace of the repo, the environment, and the home directory folder on Windows and started over.
Installed everything over again, and again tried running `l1_to_l2a_e2e.py`, which worked on WSL.
Crashed in the process of importing everything
Specifically, on line 10, `import corgidrp.walker as walker`

Traceback from Spyder:

File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File ~\documents\github\corgidrp\tests\e2e_tests\l1_to_l2a_e2e.py:10
    import corgidrp.walker as walker

  File ~\documents\github\corgidrp\corgidrp\walker.py:11
    import corgidrp.caldb as caldb

  File ~\documents\github\corgidrp\corgidrp\caldb.py:353
    default_detparams.save(filedir=corgidrp.default_cal_dir)

  File ~\documents\github\corgidrp\corgidrp\data.py:453 in save
    hdulist.writeto(self.filepath, overwrite=True)

  File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\astropy\io\fits\hdu\hdulist.py:1032 in writeto
    fileobj = _File(fileobj, mode=mode, overwrite=overwrite)

  File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\astropy\io\fits\file.py:218 in __init__
    self._open_filename(fileobj, mode, overwrite)

  File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\astropy\io\fits\file.py:651 in _open_filename
    self._file = open(self.name, IO_FITS_MODES[mode])

OSError: [Errno 22] Invalid argument: 'C:\\Users\\user\\.corgidrp\\default_calibs\\DetectorParams_2023-11-01T00:00:00.000.fits'

It stumbled on opening a FITS file from `~/.corgidrp`
Checking that directory reveals that the `default_calibs/` folder is (once again) empty

It looks like the code which populates that folder slips up on Windows and does not report any error.

Post169, 2024-10-28:
Looking more closely, it appears that the default values for calibration are successfully put together, and the problem is in the attempted file name.
ChatGPT's first guess is that the problem lies with the colons in the file name, which Windows does not allow. The recommendation is to replace them with dashes.

Post169, 2024-10-28:
Changing ':' to '-' in line 352 of caldb.py (second line of block below) is not a quick escape.
```
if not os.path.exists(os.path.join(corgidrp.default_cal_dir, "DetectorParams_2023-11-01T00:00:00.000.fits")):
    default_detparams = data.DetectorParams({}, date_valid=time.Time("2023-11-01 00:00:00", scale='utc'))
    default_detparams.save(filedir=corgidrp.default_cal_dir)
```
Error response to change:
```
  File ~\documents\github\corgidrp\corgidrp\caldb.py:352
    default_detparams = data.DetectorParams({}, date_valid=time.Time("2023-11-01 00-00-00", scale='utc'))

  File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\astropy\time\core.py:2021 in __init__
    self._init_from_vals(

  File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\astropy\time\core.py:559 in _init_from_vals
    self._time = self._get_time_fmt(

  File ~\anaconda3\envs\CorgiDRP_p312\Lib\site-packages\astropy\time\core.py:655 in _get_time_fmt
    raise ValueError(message)

ValueError: Input values did not match any of the formats where the format keyword is optional:
- 'datetime': Input values for datetime class must be datetime objects
- 'ymdhms': input must be dict or table-like
- 'iso': Time 2023-11-01 00-00-00 does not match iso format
- 'isot': Time 2023-11-01 00-00-00 does not match isot format
- 'yday': Time 2023-11-01 00-00-00 does not match yday format
- 'datetime64': Input values for datetime64 class must be datetime64 objects
- 'fits': Time 2023-11-01 00-00-00 does not match fits format
- 'byear_str': Time 2023-11-01 00-00-00 does not match byear_str format
- 'jyear_str': Time 2023-11-01 00-00-00 does not match jyear_str format
- 'astropy_time': Input values for astropy_time class must all be the same astropy Time type.
```

Clearly the datetime needs to be recognizable in a standard form.
Therefore, it appears to me at this moment that it will be necessary to add another step to each of the classes in `data`, processing the datetime values and turning them into strings that are acceptable within Windows filenames.

semaphoreP, 2024-10-29:
Thanks for the detailed investigation on this. I think this is fixable, but are you ok with using WSL?

Post169, 2024-10-29:
For now, I think I can use WSL

semaphoreP, 2024-10-29:
Ok great. I think the addition we need to do is to do a check and replace on the filename when reading in the calibration file. It's a pretty straightforward fix, when we get around to it.

Post169, 2024-11-04:
Note, the README still says, at the beginning of the first solid paragraph in the Install section, "The installation will create a configuration folder in your home directory called `.corgidrp`."
If that is out of date, there may be other parts to update as well.

maxwellmb, 2024-12-13:
@Post169 was this resolved? I'd like to close the issue or identify the next steps.

Post169, 2024-12-13:
The original issue was that I was not able to run corgidrp on Windows, but I was pointed to a successful workaround for that.
A likely fix was identified by @semaphoreP.
My last comment was that the README needed some updating.

semaphoreP, 2025-12-17:
Seems like there's a suitable workaround, so closing this issue. Please re-open if needed.

closed 2025-12-17 by semaphoreP