roman-corgi/corgidrp #229: corgidrp.data class instances must be pickle-able for the docker image to work

ajeldorado, 2024-10-25:
**Describe the bug**
Most functions and methods take as the first argument a class instance from corgidrp.data such as Dataset, Image, etc. For them to be input arguments, they have to be instantiated before the function/method call, which means that the docker image must be able to create instances of those class objects. The data objects are created with either "data_or_filepath" inputs. I get no error with direct data passed in, but I always do for the input list of filepaths when using the docker image. The error I get is `TypeError: cannot pickle 'BufferedReader' instances` when the docker image or its wrapper tries to pickle the class instance.

**To Reproduce**
This can be reproduced without the docker image by creating a class instance of say corgidrp.data.Dataset using a list of filenames as the input and then trying to pickle the object.

**Expected behavior**
The data class objects need to be pickle-able in order to use the docker image of corgidrp.

**corgidrp version:**
v1.0

**Additional context**
As Al Niessner put it, "The original agreement/pact was that all object that are passed via function arguments need to be pickle-able for dorba to work." 

I tried adding the keyword argument `memmap=False` as described [here](https://docs.astropy.org/en/latest/io/fits/appendix/faq.html#id11) in corgidrp/data.py whenever astropy.io.fits.open() is called, but that didn't fix it. I thought it might because it makes the data be loaded rather than waiting to load it, which of course can't work when you try to pickle it. I also tried setting `os.environ['MKL_NUM_THREADS'] = "1"` at the top of corgidrp/data.py to see if that would disable multithreading, but it did not change the issue with BufferedReader. So I don't know what else can be done to avoid or disable BufferedReader, but we have to figure out a way for this to work.

semaphoreP, 2024-10-25:
I've dealt with this in some other repos in the past, so can investigate. Is this critical or nice to have?

ecady-jpl, 2024-10-25:
Hi @semaphoreP, this is critical, the R3.0.1 release won't be deployable without this.  @ajeldorado is the first to run into it, but @mygouf will have this problem with her testing as well.

semaphoreP, 2024-10-25:
Ok, I think I found the issue (you can't pickle hdu_lists). A little tedious, but will turn this into dictionaries when stored as objecet fields.

al-niessner, 2024-10-25:
The best option is to make instantiation of the astropy object as lazy as possible (first read/write access from an outside caller). Hence the parent object can be instantiated and pickled (offending astropy object will still be None) and then read inside the container at first access request.

maxwellmb, 2024-10-31:
Although the PR has gone through, I'm leaving this issue open until @ajeldorado confirms his problems are solved upon return from vacation.

maxwellmb, 2024-12-13:
@ajeldorado, since we haven't heard from you again, I assume that this was successfully resolved, but it would be great to have confirmation

ajeldorado, 2024-12-13:
Yes, the problem I was having has been fixed, so go ahead and close this.

PR #230 (2024-10-31): Fix pickling for dataset
## Describe your changes

Copy hdulist after read, so file resources are released. Added unit test to check datasets read in from disk can be pickled. 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Reference any relevant issues (don't forget the #)

Closes #229 

## Checklist before requesting a review
- [x] I have linted my code
- [x] I have verified that all unit tests pass in a clean environment and added new unit tests, as needed
- [x] I have verified that all docstrings are properly formatted and added new documentation, as needed
closed 2024-12-13 by maxwellmb