roman-corgi/corgidrp #231: Write sorting algorithm for EM-gain, k-gain and non-linearity

hsergi, 2024-10-28:
This is a feature that comes as one of the to-do tasks of this parent issue about EM-gain, k-gain and non-linearity: https://github.com/roman-corgi/corgidrp/issues/226

The task is:

Write the sorting algorithm for this calibration data as a stand-alone function in DRP. 

Keep the CAR team involved in the development of the sorting algorithm. 

Make sure the algorithm is consistent with the observer’s choices

hsergi, 2024-11-04:
@semaphoreP: 
Standalone sorting function in corgidrp folder? Or inside mocks.py?

Do you have any preference for the name of the function? So far, I'm using calsort.py

Input: list of L1 filenames?
Output: list of L1 filenames for the mean frame and list of L1 frames for the calibration type?

Should the sorting function sort for all cal types when it's called, or k-gain/non-linearity or EM-gain depending on how it's called? For instance, if running a test on k-gain, it will sort all of them, or only k-gain.

When to add OBSTYPE to the FITS files and store again? This requires update the L1 headers of the many files. Should it be done in the sorting algorithm, or after, once the output files are retrieved by the algorithm. That's something that @juliamilton is working on

Technical note: I had to increase the number of files that my mac can open by default (there are ~500 files) ulimit -n 500. Not sure how this will affect the auto testing and other users.

semaphoreP, 2024-11-04:
Location: I would put it into a standalone file like `sorting.py`, or inside either the calibrate_kgain or calibrate_nonlin python files.

Input: dataset
Output: dataset that downselects to the set of data you want (kgain, nonlin, em-gain)

Sorting function should thus downselect to the files needed for the requested calibration dataset (not all of them). I envision this as the first step of the kgain/nonlin recipe (basically taking the whole pupil imaging dataset, and down selecting).

I think it makes the most sense for your function to add the OBSTYPE back into the header.

The open file limit is a common mac-specific issue. CTC runs on Linux, which defaults to a much higher open file limit.

hsergi, 2024-11-04:
Location: this sorting function is also to be used by em-gain. Other functions that were in k-gain and used by non-linearity, they ended up in mocks.py. I would write it for now as a stand alone function. If we want to add it inside one of the calibration functions, we can do that

Input: how to know which calibration type needs be sorted if only a dataset is the input? 
Yes, it needs to take the whole pupil imaging dataset, and down select

semaphoreP, 2024-11-04:
It shouldn't be inside another function, but it can be another function inside one of the calibration modules for either kgain or nonlin. Alternatively, it's fine to put it in it's own module/python file. It shouldn't be inside mocks.py (mocks is only used for simulating test data, not for processing real data).

The input should also have a keyword argument that's a string that the user needs to specify what kind of dataset they want ('kgain', 'nonlin', 'emgain').

juliamilton, 2024-11-04:
I agree with Jason, makes sense to me to add OBSTYPE into the L1 header in the sorting function when a frame has been identified and tagged as being for mean frame, k-gain, non-lin, etc.

semaphoreP, 2024-11-08:
Finished in PR #247

closed 2024-11-08 by semaphoreP