roman-corgi/corgidrp #233: Alternative way to directly parse a sorted list of frames for each calibration type

hsergi, 2024-10-28:
This is a feature that comes as one of the to-do tasks of this parent issue about EM-gain, k-gain and non-linearity: https://github.com/roman-corgi/corgidrp/issues/226

The task is:

Include an alternative way to directly parse a sorted list of frames for each calibration type. 

Marie has decided that I would first work on implementing the sorting algorithm and its unit tests: https://github.com/roman-corgi/corgidrp/issues/231 and https://github.com/roman-corgi/corgidrp/issues/232

semaphoreP, 2024-10-29:
Can't we do this manually, with a recipe that species each individual file we want to include in the input dataset, and in the order we want them? 

Or is this specifically to handling it in the EM-gain/kgain/non-lin code?

hsergi, 2024-10-29:
This is specifically for the EM-gain/kgain/non-lin code. You (Jason) suggested this idea. Maybe to keep some flexibility, and also in case some constraints change and the sorting algo is not adequate (?). I'm fine with whichever way you choose. Let me know

semaphoreP, 2024-10-30:
Sure, I agree flexibility is good. I'm just not sure what new we need to add here for this.

hsergi, 2024-10-30:
I think your suggestion is all what is needed: "a recipe that species each individual file we want to include in the input dataset, and in the order we want them" 

This way we can easily cover edge cases, like a few frames needed to be retaken due to some issue. With this recipe-alternative, the sorting algorithm can be designed to comply with the default rules that the observer will use when writing the detailed collection of calibration data (contiguity, time-ordered, pairing rules, ...) that I have collected so far from Kevin and Guillermo (who's working with Bijan and Peter Williams on this too)

I'll implement then this option in the sorting function. Once it is done and tested, we can close this issue.

Let me know if you have any further suggestions

semaphoreP, 2024-10-30:
Just to clarify I'm thinking the sorting function is a step function we apply to filter down the dataset into just the relevant files. This recipe file solution would just manually list the files in a custom recipe, and process that custom recipe. We would basically skip the sorting function in the manual list case.

hsergi, 2024-10-30:
I had considered it as an optional argument to the sorting function. As far as there is a clear way on how to choose each option (instead of parsing a non-None argument in the sorting function), I agree with you that it can be outside the sorting function

semaphoreP, 2024-12-12:
I'm coming back to this, and not sure there's anything actionable here that's not already in the DRP. Please re-open, if there is something action-able to do.

semaphoreP, 2024-12-12:
You would just make the recipe and write out each individual input file into the `inputs` field of the JSON file. 

I don't think we need UTs because other ones already cover this functionality.

hsergi, 2024-12-12:
Do you mean that the field ""inputs" : []," in the recipe will automatically trigger the creation of a dataset that will be the input for the steps?

If so, we can close this issue, in the sense that there's a definite way to create a dataset provided by ""inputs" : []," and it has been tested

semaphoreP, 2024-12-12:
Yes, that's right. That's how the walker creates the first dataset when processing a recipe.

closed 2024-12-15 by hsergi