roman-corgi/corgidrp #237: Compute apparent magnitude of source in CGI band

semaphoreP, 2024-10-29:
**Is your feature request related to a problem? Please describe.**
Given the magnitude of the star in a standard filter band and the star's spectra type, compute the apparent magnitude of the source in the CGI band and store it in the header

**Describe the solution you'd like**
The above in a step function. 

**Additional context**
Need to figure out where the CGI filter profiles are, but they should already be documented

JuergenSchreiber, 2024-11-13:
I guess you mean something like a color correction factor?

semaphoreP, 2024-11-13:
Yep, exactly it.

semaphoreP, 2024-12-05:
@JuergenSchreiber, I was just looking over this issue more closely, and I'm not sure #259 fully solves this issue, because we computed a color correction, but we didn't go all the way and compute the apparent magnitude in the CGI band. Is that correct?

JuergenSchreiber, 2024-12-06:
Well, we will compute the flux in the CGI bands, and I think we could convert the units to magnitudes. But that might not be trivial since we do not have the standard UBV filters. Since the goal is the calibration of the ND filter, so relative fluxes is good enough, I think the final unit is not that important as long as we stay at the same all the time. Is this reasonable?

semaphoreP, 2024-12-06:
I think we want vega magnitudes, so given a filter profile and a vega spectrum, it should be doable.

I only emphasized magnitudes, because that's what's listed in the design document. I think for the science data, we do want to be able to calculate the apparent magnitude of both the star and any point sources around it (not just the flux ratio). So we do need some way to convert to fluxes (not just relative fluxes to the star).

JuergenSchreiber, 2024-12-09:
Ok, so please keep this issue open, I will write it in a header keyword

closed 2024-12-10 by semaphoreP