roman-corgi/corgidrp #240: Measure properties of detected companions in PSF-subtracted coronagraphic images

semaphoreP, 2024-10-29:
**Is your feature request related to a problem? Please describe.**
Want to measure the position, flux ratio, and apparent magnitude of any point sources in a coronagraphic image

**Describe the solution you'd like**
A step function, given a list of detected sources listed in the header, that measures precise positions and fluxes (may require forward modeling, may be able to get away with regular PSF fitting for RDI)

**Additional context**
Could use code from pyklip/spaceklip as a start

maxwellmb, 2025-03-12:
Linking Issue #328, because there was useful discussion there. 

@juliamilton, I think you can assume that the APP_MAG header keyword is already populated, but to implement it as @semaphoreP has described I think you may need to assume that there's an off-axis stellar image in the dataset and we'll have to be able to identify that.

semaphoreP, 2025-03-17:
I would say let's get something working, and then we add contingencies? We could probably produce a calspec file for each of our stars given that we won't be looking at very many stars.

semaphoreP, 2025-03-27:
implementation complete in #364 with the acknowledgement we can improve it (See #377)

closed 2025-03-27 by semaphoreP